roleoroleo/yi-hack-Allwinner-v2 #581: Multimedia on Home Assistant Error

hiagocosta, 2023-02-16:
Hi, when I try to navigate into the recorded videos inside the camera with multimedia menu in home assistant, it gives me this error 

[image]

<html>
<body>

Firmware Version | 0.2.5
-- | --
Base Version | 9.0.35.00_202111291330
Model Suffix | y291ga

</body>
</html>

roleoroleo, 2023-02-16:
HA version?
Integration version?

hiagocosta, 2023-02-16:
[image]
[image]

roleoroleo, 2023-02-18:
I think the problem is already fixed but not released.

hiagocosta, 2023-02-24:
After update The CAM still keeps in that situation,

I see The logs and I have this:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/media_source/__init__.py:173
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19:27:20 (4 occurrences)
Last logged: 19:29:08

  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/media_source/__init__.py", line 173, in websocket_browse_media
    media.as_dict(),
AttributeError: 'NoneType' object has no attribute 'as_dict'

roleoroleo, 2023-02-25:
Are there errors related to my integrations?
Are you using a special network configuration?
Reverse proxy, firewalls, etc...?

closed 2023-06-25 by stale[bot]