roleoroleo/yi-hack-Allwinner-v2 #585: Issue with Yi Home Y4 1296P Fullhan chip

redphx, 2023-02-25:
I bougth a new Y4 camera from China today, and I think it's a new version of Y4

SN: AKY4CH... (not IFCN)
Model: YYS.2919 (I think it's still the same as other versions)
Chip: FH8632V2
Firmware: 5.1.00.01_202212281036 (latest version)
Not region locked. Was able to pair it using the Yi Home China app.

I tried to hack it with the y29ga file multiple times but it didn't work. Then some how it was able to create a new boot partition, then it stopped. I tried to format and run multiple times later and never got it working again.  

The LED light stucks at dimmer blue light, instead of bright blue.

I got this file when browsing the SD card: [wifi_connected.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10831381/wifi_connected.zip)
  
[image]
[image]

redphx, 2023-02-25:
Surprisingly I can download firmware files by just opening its FTP server. Do you want these files?

[image]

redphx, 2023-02-26:
I was able to dump the `/backup/` folder by running the `debug.sh` file in the SD card.  Not sure why it didn't run `Factory/config.sh`. Maybe it's missing chmod permission?
  
Hope these files are useful:
- [backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10832551/backup.zip)  
- [ftp.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10832550/ftp.zip)

roleoroleo, 2023-02-26:
At the moment, I can't support it.
I should buy one.

redphx, 2023-05-29:
comment to prevent closing

closed 2023-09-26 by stale[bot]