roleoroleo/yi-hack-Allwinner-v2 #587: Not working on Y28GA

graham30rad, 2023-02-28:
Hello, I have placed the files from the folder to the SD as required, but no luck, the camera will turn on and stream to the app, but no RTSP. I have tried this many times but no joy. 

here is the log if it helps

tmp/sd/first_login/                                                                                 0000755 0000000 0000000 00000000000 14377437674 013327  5                                                                                                    ustar   root                            root                                                                                                                                                                                                                   tmp/sd/first_login/ps1.txt                                                                          0000755 0000000 0000000 00000004601 14377437674 014577  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                     PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  207 root         0 SW   [oom_reaper]
  208 root         0 SW<  [writeback]
  210 root         0 SW   [kcompactd0]
  211 root         0 SW<  [crypto]
  212 root         0 SW<  [bioset]
  214 root         0 SW<  [kblockd]
  255 root         0 SW   [sys_user]
  262 root         0 SW   [kworker/0:1]
  263 root         0 SW<  [cfg80211]
  269 root         0 SW<  [watchdogd]
  284 root         0 SW<  [spi0]
  294 root         0 SW   [kswapd0]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  388 root         0 SW<  [bioset]
  393 root         0 SW<  [bioset]
  398 root         0 SW<  [bioset]
  403 root         0 SW<  [bioset]
  408 root         0 SW<  [bioset]
  413 root         0 SW<  [bioset]
  418 root         0 SW<  [bioset]
  450 root         0 SW   [irq/302-sunxi-m]
  452 root         0 SW   [irq/166-sdc0 cd]
  453 root         0 SW   [irq/303-sunxi-m]
  456 root         0 SW   [kworker/0:2]
  488 root         0 SW   [kworker/u2:2]
  509 root         0 SW<  [bioset]
  510 root         0 SW   [mmcqd/0]
  531 root         0 SW<  [kworker/0:1H]
  544 root         0 SWN  [jffs2_gcd_mtd4]
  551 root       980 S    -/bin/sh
  583 root       980 S    /usr/sbin/telnetd
  586 root         0 SW   [kworker/0:3]
  594 root         0 SW   [ksdioirqd/mmc1]
  600 root         0 SW   [RTW_XMIT_THREAD]
  601 root         0 SW   [RTW_CMD_THREAD]
  602 root         0 SW   [RTWHALXT]
  642 root      1088 S    ./dispatch
  648 root     38924 S    ./rmm
  720 root      1812 S    ./mp4record
  721 root      1148 S    ./cloud
  722 root      2984 S    ./p2p_tnp
  723 root      3188 S    ./oss
  725 root       676 S    ./watch_process
  732 root      1400 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
  764 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
  860 root       980 S    sh -c rm /tmp/sd/log_first_login.tar.gz;rm /tmp/sd/f
  874 root       980 R    ps
                                                                                                                               tmp/sd/first_login/homever                                                                          0000755 0000000 0000000 00000000027 14377437674 014721  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   9.0.20.06_202007061841
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         tmp/sd/first_login/lsmod.txt                                                                        0000755 0000000 0000000 00000001043 14377437674 015207  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   Module                  Size  Used by
pid_list                1389  1 
ssp_ms41909             4024  2 
cpld_periph             5329  2 
[... 209 lines cut ...]
-rwxrwxr-x    1 root     root        114344 Jul  6  2020 libaw_aacdec.so
-rwxr-xr-x    1 root     root         83420 Jul  6  2020 libcdx_base.so
-rwxr-xr-x    1 root     root         25848 Jul  6  2020 libcutils.so
-rwxr-xr-x    1 root     root         21588 Jul  6  2020 libcdx_common.so
-rwxr-xr-x    1 root     root         13604 Jul  6  2020 liblog.so
-rwxr-xr-x    1 root     root         13396 Jul  6  2020 libion.so
-rwxr-xr-x    1 root     root          9316 Jul  6  2020 libaw_g711adec.so
-rwxr-xr-x    1 root     root          9316 Jul  6  2020 libaw_g711udec.so
drwxrwxr-x    2 root     root           188 Jul  6  2020 .
drwxrwxr-x    3 root     root            40 Jul  6  2020 ..
                                                                                                                                                tmp/sd/first_login/wpa_supplicant.conf                                                              0000755 0000000 0000000 00000000255 14377437674 017234  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="XXXXXXXX"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="XXXXXXX"
}
                                                                                                                                                                                                                                                                                                                                                   tmp/sd/first_login/df.txt                                                                           0000755 0000000 0000000 00000000376 14377437674 014472  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/mtdblock3            2816      2816         0 100% /home
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1280      1280         0 100% /

graham30rad, 2023-02-28:
I pulled off the cover and the board said Y29GA, not Y28GA, so I installed the y28ga, still the same issue, only now it's not even showing in the app .. the yellow light comes on, after about 15/20 ish seconds. it goes off briefly and back on again for another 10 seconds, then stays off. My router is seeing it, and it's IP is pinging.  http://"IP-ADDRESS":8080 does not work either.

graham30rad, 2023-03-02:
Can anybody help me out with this??

roleoroleo, 2023-03-04:
As described in the readme page, you have to check the fw version to identify your cam.

graham30rad, 2023-03-04:
Yes, I did, the firmware on mine is 9.0.20* . . which is the y28ga on your list, but the issue I have is the model on the sticker on the back of the cam AND on the box says y28ga . . BUT, on the actual circuit board inside it says y29ga. I have tried both "hacks" but none of them work, does this mean that my particular camera can not be modified then?? Also in the instructions it says "Download the latest release from the Releases page." . . is there a releases page that I can't see?? And point #3 says "Extract the contents of the archive to the root of your SD card." but if I do that the sd card does not look like what is required 
|-- Factory/
|-- yi-hack/
|-- lower_half_init.sh
so I have to place the files manually. The instructions are not that clear. I really appreciate what you've done, and thank you for the modified files in the past which are running on my other camera perfectly, I just wanted to know if I was doing something wrong is all, or if the camera is not compatible.
Regards.

roleoroleo, 2023-03-05:
If the fw release is 8.0.20 your cam is a y28ga.
Don't worry about labels on the cam.

This is the release page: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases
Download this archive and extract it in the sd card:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.6/y28ga_0.2.6.tgz

graham30rad, 2023-03-05:
Yes that works, thank you very much! The files that I had downloaded first looked nothing like those, not sure why . . but all is good now. Thanks again, and have a great day! 
Regards.

closed 2023-07-09 by stale[bot]