roleoroleo/yi-hack-Allwinner-v2 #589: The image cannot be displayed because it contains errors.

pwlgrzs, 2023-03-02:
Firmware Version | 0.2.6
Base Version | 9.0.21.08_202007270941
Model Suffix | h51ga

Issue:
Going to any of low/res snapshot links ends with: "The image cannot be displayed because it contains errors."

Camera is functional as I can access ONVIF stream, going to snapshot link do not reset camera, just throws error.

cjgarciao, 2023-03-04:
I had a similar issue with h30ga after update yi-hack to 0.2.6. I have deleted all events and, I don't know why, but now works better.
Carlos.

roleoroleo, 2023-03-04:
Normally this happens when another snapshot is currently running.
How many clients do you use?
Web ui? HA?

pwlgrzs, 2023-03-05:
Only client is Synology on ONVIF stream. Nothing is accessing snapshots.

I have removed all events and rebooted, but this had no effect. Same applies for disabling streaming on Synology side.

roleoroleo, 2023-03-06:
Try to disable this option:
[image]

closed 2023-07-09 by stale[bot]