roleoroleo/yi-hack-Allwinner-v2 #592: Outdoor Camera - RTSP stream unavailable

fibilt, 2023-03-03:
Firmware Version 0.2.6
Base Version 11.1.26.1_202301191526
Model Suffix h30ga

RTSP stream is very unreliable after updating to 0.2.6 and 11.1.26.1. I have noticed that very frequently, the camera gets unresponsive for some time when trying to stream video via RTSP, and then it recovers automatically from it.

Integration with home assistant does not work either after the update. When I try to stream video, I get the following error in HA:

2023-03-03 16:18:37.410 ERROR (SyncWorker_1) [custom_components.yi_hack.camera] Fetch snapshot image failed from yi_hack_a2_0efa89_cam, falling back to FFmpeg; HTTPConnectionPool(host='192.168.145.122', port=8080): Read timed out.

motion detection cam snapshot does not work either.

[image]

Is this a known bug of 0.2.6 version? Can someone with same camera confirm if they dont have any problem with 0.2.6 version please?

fibilt, 2023-03-06:
@roleoroleo any thought on what can be causing this issue?

roleoroleo, 2023-03-07:
Maybe the software update did not complete successfully.
Please, try to reinstall the hack.

fibilt, 2023-03-07:
Hi @roleoroleo! 

I have reinstalled the hack and also tried to use a new sd card (Samsung evo 32gb) with the same results. For some reason, the integration with HA does not work.

These are some of the errors I can see in HA logs:

- 2023-03-07 09:34:54.943 ERROR (stream_worker) [homeassistant.components.stream.stream.camera.yi_hack_a2_0efa89_cam] Error from stream worker: Error opening stream (INVALIDDATA, Invalid data found when processing input) rtsp://192.168.145.122/ch0_0.h264
- 2023-03-07 09:35:00.371 ERROR (SyncWorker_6) [custom_components.yi_hack.camera] Fetch snapshot image failed from yi_hack_a2_0efa89_cam, falling back to FFmpeg; HTTPConnectionPool(host='192.168.145.122', port=8080): Read timed out.
- 2023-03-07 09:35:09.955 ERROR (stream_worker) [homeassistant.components.stream.stream.camera.yi_hack_a2_0efa89_cam] Error from stream worker: Error opening stream (ERRORTYPE_110, Operation timed out) rtsp://192.168.145.122/ch0_0.h264

As I said, I can stream video via rtsp (high and low) from VLC on my Mac or iPhone, but not from yi_hack intregation. When I click on camera  icon to stream video, the status is on, but right after clicking on the icon status changes to "unavailable". Then after some minutes comes back to on, but same behaviour every time I try to stream video from yi_hack integration.

Are any logs being captured in the camera itself? I can ssh into it and provide further information if that helps.

Thanks beforehand!

fibilt, 2023-03-09:
Thanks @dreimer1986 for your comments.

I used webrtc but with a slightly different approach, and it works great. Thanks for pointing me to this integration. This is what I did just in case you want to give it a try:

type: custom:webrtc-camera
url: rtsp://192.168.145.122/ch0_0.h264

Instead of using the yi_hack integration entity, I used the rtsp stream directly, and it works like a charm. I dont need to install the yi_hack integration for this, which does not work for me.

@roleoroleo will you be able to look into the integration issue at some point?

roleoroleo, 2023-03-09:
I don't understand the problem with the stream.
yi_hack_integration doesn't provide stream.
The rtsp stream is provided directly by the cam.

About motion detection and other sensors, please try to remove the cam from ha and add it again.

fibilt, 2023-03-10:
@roleoroleo this is what I am saying. That RTSP stream provided by the camera works if I use with VLC for instance. But I cant stream video through yi_hack integration. I could before the update though.

roleoroleo, 2023-03-10:
Never used this type of control.
But tested with my cam and it works.
Did you try to remove from and install it again?
Or try with webrtc: https://github.com/AlexxIT/WebRTC/

roleoroleo, 2023-03-10:
If it doesn't work, I need help to debug this model because I don't own it.

Miguelin96, 2023-03-13:
I have the same problem, the camera is blocked a few minutes after accessing with the plug-in to HA.

roleoroleo, 2023-03-13:
Please, check if swap file is enabled.
Enabling snapshot with swap file disabled causes reboots on some models.

fibilt, 2023-03-13:
Thanks @roleoroleo for your comments. In my case, swap is disabled. I am happy to help you debug this model. Just let me know how I can help.

fibilt, 2023-03-15:
thanks @roleoroleo. Enabling swap, video stream from the cam sensor of yihack_integration works. 

However, the high-resolution snapshot still does not work. I can open the low resolution snapshot but not the high res one.

roleoroleo, 2023-03-15:
Where do you open high res snapshot?

fibilt, 2023-03-16:
I use the URL:

[image]

roleoroleo, 2023-03-16:
Never?

fibilt, 2023-03-16:
Yes, never. It used to work with 0.2.5 hack but since I updated to 0.2.6 I never got the high res snapshot to work.

roleoroleo, 2023-03-16:
Login with ssh and try to enter the command manually:
`imggrabber -m h30ga -r high -w -d > /tmp/sd/temp.jpg`

fibilt, 2023-03-16:
wit this command it works. The image was saved and I scp it to my Mac and open it without any problem.

fibilt, 2023-03-16:
The only bug I see now is that when video is recorded after motion detected, 2 videos are always stored. One capturing the motion, and a second one recording a few seconds after the motion was detected.

[image]

[image]

roleoroleo, 2023-03-16:
> The only bug I face now is when a video is recorded after motion is detected, two videos are always stored. One with the motion captured, and a second one recorded a few seconds after the motion was detected.

This is ok, the max length of a video is 1 minute.
I the event lasts 2 minutes, you will find 2 videos.

closed 2023-07-14 by stale[bot]