roleoroleo/yi-hack-Allwinner-v2 #593: QG2101B firmware 9.0.29.99_202012051744 use b091qp cannot connect to wifi and rtsp

salvomaltese, 2023-03-06:
Hi,

I've the same problem of https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/335

This is my factory firmware and boot log and ps
https://drive.google.com/drive/folders/1jq1g4fgt_o1LJiX3p9VSO1J2fdGhqCWY?usp=share_link

roleoroleo, 2023-03-06:
It should be already supported.
The model suffix is q321br_lsx:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.6/q321br_lsx_0.2.6.tgz

roleoroleo, 2023-03-06:
What's the name of the cam?

salvomaltese, 2023-03-07:
> 

Model camera is: IHOUMI D350

salvomaltese, 2023-03-07:
EN: before writing here I had installed the version b091qp, is it possible to flash again by installing the version you indicated?
Thanks so much.

IT: prima di scrivere qua avevo installato la versione b091qp, è possibile rieseguire il flash installando la versione che mi hai indicato?

roleoroleo, 2023-03-08:
You can simply edit the file:
/tmp/sd/yi-hack/model_suffix
and change b091qp with q321br_lsx

salvomaltese, 2023-03-08:
Undue /tmp has present only topfile
root@(none):/# cd /tmp
root@(none):/tmp# ls
topfile

salvomaltese, 2023-03-08:
i performed the following steps:
1 - followed successfully the unbricking guide https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam
2 - Re-run the first configuration with YI IOT app
3 - applied the FW [q321br_lsx](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.6/q321br_lsx_0.2.6.tgz) 

But I have the same problem

salvomaltese, 2023-03-08:
After many attempts I realized that the problem was not enough disk space on /backup mount.
I have modify config.sh and add this command:
rm -f  /backup/tools/rsa_pub_dec

Now Works
[image]

Low res snapshot OK
High res stream OK (verified on VLC)
Only audio stream OK (verified on VLC)

High res snapshot go in Out of memory and cam reboot:
[  211.203679] Out of memory: Kill process 1187 (rmm) score 530 or sacrifice child
[  211.219811] Killed process 1187 (rmm) total-vm:49832kB, anon-rss:4888kB, file-rss:27776kB, shmem-rss:536kB

closed 2023-03-08 by salvomaltese

roleoroleo, 2023-03-09:
> After many attempts I realized that the problem was not enough disk space on /backup mount.

This problem can be fixed (afaik) creating the restore partition using "hacked" parameter.

> High res snapshot go in Out of memory and cam reboot:

Enable swap file.