roleoroleo/yi-hack-Allwinner-v2 #594: y211ga: Installed FW 12 and Hack 0.2.6

dreimer1986, 2023-03-07:
After FW 12.0.37.0_202211241030 installing by official app and 0.2.6 hack reinstalled the camera snapshotting does not work anymore in HA. In my case trying to access the camera entity in any way at all (for exmaple for the snapshot making) causes this "error":

[image]

Created snapshots are 0 Bytes big.

The stream over WebRTC Addin works fine though.

I am very sure that I can get any useful logs off HA to support here, but as always I have NO idea which logs I should set to "debug". In this case HA is a mess IMO.

dieideeistgut, 2023-03-07:
Try re-running the thumbs.sh - since 0.2.6. i have many failures in the thumb-generation somehow. still looking into this.

If i am not misaken for y211ga its:

. /tmp/sd/yi-hack/scripts/env.sh
/tmp/sd/yi-hack/scripts/thumbs.sh stop
/tmp/sd/yi-hack/scripts/thumbs.sh cron

Watch the output for FAILURE. Delete the .mp4. Re-Run. This usually fixes this currently for me.

dreimer1986, 2023-03-07:
Thanks for reply.

The HA service uses the camera entity aka the video stream and creates a picture off it. As you can see above this fails because the camera does not really show anything outside of WebRTC custom addin. So I am quite sure that this is not the fix.

I can confirm that the rtsp stream works fine on VLC, but somehow HA can not work with it at all anymore. I have no idea if the addin is the big problem here or the stream from the camera though... I updated to the most recent master version of the addin, but the added fixes are not helping in this case. Ah and while rebooting I got this:

[image]

The cameras are very ON though:

[image]

dreimer1986, 2023-03-07:
OK, forget it. I checked the addin's settings for both cams and both were "off" O_o Now they work fine. Snapshots do not, the HA made ones, but the cam makes fine ones. So all left is likely HA's bugs and thus this thing can be closed. Motion detection does not really work, too, but this one is for a later time and another report.

closed 2023-03-07 by dreimer1986