roleoroleo/yi-hack-Allwinner-v2 #595: y211ga: Installed FW 12 and Hack 0.2.6, motion detection not working

dreimer1986, 2023-03-07:
After FW 12.0.37.0_202211241030 installing by official app and 0.2.6 hack reinstalled the camera does not really allow any motion detection anymore (MQTT in HA). Even more interesting, in camera settings the switch for motion detection always switches back to off after I refresh the page.

[image]

Trying the same over REST link http://x.x.x.x:8080/cgi-bin/camera_settings.sh?motion_detection=yes brings back this:

```
{
"error":"false"
}
```

This happens on two cams, not just one, so I think that a problem on flashing is not our problem here...

roleoroleo, 2023-03-08:
Check this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/b40b880afa217446582eeaa6664541f661ccb59c

dreimer1986, 2023-03-08:
Ooops, I thought this one was in 0.2.6 already. OK, then this is fine as it's already fixed upstream.
Regarding the motion detection failing. After I switched on the cam on Addin side and it was active on both sides (cam and ha) it did not work... Now a few hours later it indeed sends images again. I think it takes a while to start working maybe?
Regardless, I think it's working as expected now.

closed 2023-03-09 by dreimer1986