roleoroleo/yi-hack-Allwinner-v2 #596: Yi Dome Guard RFUS - Reboots with 0.2.6

auanasgheps, 2023-03-08:
Hi,

Since I updated to 0.2.6 this camera reboots spontaneously every now and then.
Model Suffix is r35gb

here is a series of reboots, reported by the HA integration.
[image]

I do not know what logs should I check, please let me know.
Also, when it reboots spontaneously, the camera is pointed to another direction.

roleoroleo, 2023-03-09:
Please, try to disable these options:
[image]

auanasgheps, 2023-03-09:
Yup they are already disabled.

[image]

auanasgheps, 2023-03-11:
Rebooted again :(

roleoroleo, 2023-03-11:
What's the last running hack?

auanasgheps, 2023-03-12:
I had not downgraded yet, so I followed your instructions and rebooted the camera.

It may take a couple of days for the crash to occur. When it happens, what files should I grab?

auanasgheps, 2023-03-12:
After I made the changes, the camera has rebooted 2/4 times before stabilizing.

roleoroleo, 2023-03-12:
Please, send me the file /tmp/sd/yi-hack/etc/system.conf
and share your system.conf, if you can (remove personal data).

auanasgheps, 2023-03-12:
```
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_ALT=no
RTSP_STREAM=high
RTSP_AUDIO=
SPEAKER_AUDIO=yes
SNAPSHOT=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_NETIF=wlan0
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=30
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=<redacted>
CRONTAB=
SNAPSHOT_VIDEO=no
SNAPSHOT_LOW=no
DEBUG_LOG=yes

```

roleoroleo, 2023-03-12:
> Please, send me the file /tmp/sd/yi-hack/etc/system.conf

Sorry, my mistake.
Please, send me the file /tmp/sd/hack_debug.log

auanasgheps, 2023-03-12:
Here it is! [hack_debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10952392/hack_debug.log)

roleoroleo, 2023-03-13:
The log is perfect.
Compared to my log you have more free memory than me.
The only difference is that I don't use mqtt_adv.

auanasgheps, 2023-03-17:
I've checked the camera and it rebooted spontaneously two days ago.

`Uptime | 2d 8h 19m 14s`

It keeps going on forever I believe.

RAM is not an issue: Free/Total Memory | 33244/60676 KB

roleoroleo, 2023-03-18:
Maybe a power issue?

auanasgheps, 2023-03-20:
The camera, along with other devices, is connected to a UPS.

It's running stable so far - 4d 18h of uptime, but I haven't touched a single option since I flashed the update, so I cannot explain the instability. I'll keep you posted

lolorc, 2023-03-22:
same here with r35gb / 0.2.6, I'll check the power, it's happening here every 8-12h.

roleoroleo, 2023-03-23:
If we exclude power issues and hardware failure, there are 2 reasons why the camera can reboot:
1 - Out of memory: the kernel kills unnecessary process and finally the cam reboots.
Solution: enable swap file.
2 - The process rmm dies and the watchdog restart the cam.
Solution: I don't know...
I could try to log the output of the process and check for errors.

roleoroleo, 2023-03-23:
Does 0.2.5 work properly?

auanasgheps, 2023-03-23:
I would like to downgrade to 0.2.5, but I believe the latest HA integration won't work. 
It's a problem because I have a number of automation that rely on this camera's sensors and I spent a lot of time to adapt them to the new integration,

lolorc, 2023-03-24:
looks like it was a power issue here.
I guess I could disable check_rmm in /tmp/sd/yi-hack/script/wd_rtsp.sh to see if it's a  rmm issue, kill wd_rtsp.sh or make it  log a message before rebooting.
I think I'll kill it because I don't like the while true; loop ;-)

roleoroleo, 2023-03-24:
Yes, try to kill the watchdog (or disable it from system.sh) and check if the cam remains on.
If yes, ssh should work and you should able to login and run some command.
For example a ps to show the running processes.
And:
```
cd /home/app
./rmm
```
to check what goes wrong in rmm.

lolorc, 2023-04-03:
hi, just to let you know, it hasn't crashed with a proper power supply:
```
root@yidome:~# awk '{printf "%f\n",$1/86400}' /proc/uptime 
7.804881
```

closed 2023-04-04 by auanasgheps