roleoroleo/yi-hack-Allwinner-v2 #598: Few days after upgrade camera is offline

rt400, 2023-03-12:
I have the issue... no ssh no web no rtsp and onvif
all I have is that telnet is work and login.
The camera is installed in a high ceiling so I have an accessibility problem at the moment and I'm trying to solve it without climbing a ladder.
any idea ?

roleoroleo, 2023-03-12:
If telnet is the only workng service, probably the sd card is damaged.
You should try to login with telnet and check if the sd is properly mounted.
```
root@yi-hack-allwinner-v2:~# mount | grep /tmp/sd
/dev/mmcblk0p1 on /tmp/sd type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```
If it's not mounted you could try a fsck to restore it (if it's possible).
If it's mounted, read the content:
`ls -laR /tmp/sd`

rt400, 2023-03-12:
is this a single command?

rt400, 2023-03-12:
can you please write me the commands?  cause i am newbie in linux

rt400, 2023-03-12:
this what i have in sd folder 
[image]
 and dont have mmcblk0p1 only mmcblk0

roleoroleo, 2023-03-12:
A single command:
`mount | grep /tmp/sd`

If mmcblk0 is absent, probably the sd card is corrupted.

rt400, 2023-03-12:
[image]
this what i have in /dev folder

roleoroleo, 2023-03-13:
Try mount command:
`mount | grep /tmp/sd`

rt400, 2023-03-13:
The sdcard was defected so i put it out and format it and copy the hack
and still no web ssh only telnet
if i run this command that what i get : 
[image]

rt400, 2023-03-13:
Now the telnet connect and after few seconds disconnect.
I try to remove the hack but i can't see the correct zip .
my camera is : Yi Doom Guard QFUS 9v firmware so its : r35gb.
so any idea how to continue from here ?

rt400, 2023-03-13:
[image]
i see in folder /tmp/sd/yi-hack a file fw_upgarde_in_progress
that mean that the hack still in progress ?

rt400, 2023-03-13:
Now if i run the command : `root@(none):/tmp/sd/yi-hack/script# sh system.sh` the camera work.
But if i reboot the camera the issue return...

roleoroleo, 2023-03-13:
> The sdcard was defected so i put it out and format it and copy the hack and still no web ssh only telnet if i run this command that what i get : [image]

This situation can happen but it's not ok. The sd card is not properly formatted and there are no partitions inside the card.
I don't know if this is the problem but you should reformat the sd card: try with a linux machine following this guide:
https://www.nayab.xyz/linux-tools/partitioning-using-fdisk
Or try with a windows machine.

At the end of the procedure you should have a partition mmcblk0p1 inside the disk mmcblk0 and the mount should refer to mmcblk0p1.

roleoroleo, 2023-03-13:
Damn...
Are you able to provide me a temporary telnet access to the cam (if you want)?

rt400, 2023-03-13:
can you send me your email and i send you the link?

rt400, 2023-03-13:
maybe you have the remove hack for my cam?
i want to try to remove the hack and install it again

roleoroleo, 2023-03-14:
roleo@iperbole.bologna.it

Before making changes I would like to understand the reason.
If we need to remove the hack I'll tell you.

roleoroleo, 2023-03-14:
Use this file to unbrick the cam.
Decompress it in the sd card and start the cam.
[backup_r35gb.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10973527/backup_r35gb.gz)

closed 2023-07-13 by stale[bot]