roleoroleo/yi-hack-Allwinner-v2 #599: thumb.sh and cron-timings findings leading to minimp4

dieideeistgut, 2023-03-13:
Heya, this is just a feedback from some findings with the latest versions, ha-integration and the media thumbs. i had been digging around an issue i had (with HA) for a while now and this are my findings:

- the thumbs.sh script runs quite often. on two of my cams (allwinner-v2s) this caused some issues:
- Failed creations sometimes kept a 0-byte JPG file. I worked around this out by modding my thumb.sh to delete the 0-byte JPG  in the current run, forcing a re-creation next time. 
- slowing down the * * * * cron-pattern (disabled it for testing, running the thumbs.sh manually for now) i could reduce my missing thumbs. 

I've tested this with a 4-GB-full SD and a smaller video collection. 

Looking after the reason for the failed creations though, i stumbled upon an issue where minimp4 creates kind of empty (aka frameless) .h26x files. The .JPG creation then finds nothing. This seems to happen only on short records.

roleoroleo, 2023-03-14:
Thany you.
Please create a PR, I will integrate your code.

dieideeistgut, 2023-03-14:
Will do. What's your preferred location for the placeholder image? i have mine currently in /tmp/sd/yi-hack/fallback.jpg

roleoroleo, 2023-03-14:
/tmp/sd/yi-hack/etc/fallback.jpg

closed 2023-03-27 by dieideeistgut