roleoroleo/yi-hack-Allwinner-v2 #601: Panoramic Webcam

lanfur, 2023-03-15:
Hi everyone,
I really appreciate this hack and I wanted to ask if it was possible to evaluate the possibility of adding an automatic photo capture function every 5/10 minutes, in order to use the YI camera as a panoramic webcam via FTP. Thank you

roleoroleo, 2023-03-15:
What do you mean?
A snapshot schedule, a panoramic picture using pan or both?

lanfur, 2023-03-15:
I mean a simple snapshot function, every 5 minutes, which saves the webcam.jpg file to an ftp server
such as the function that reolink IP webcams have

roleoroleo, 2023-03-15:
I could add this feature but you could already do this today by adding a script.
At the end of the boot procedure, the cam runs a script in the sd card.
You could use this script to make a snapshot and send it to a server.
And schedule it with crond.

lanfur, 2023-03-15:
thank you for your interest in my proposal
about the script, I'm not able to do it.. I don't have programming skills.. I'm a simple user

lanfur, 2023-03-15:
thank you sincerely

closed 2023-07-13 by stale[bot]