roleoroleo/yi-hack-Allwinner-v2 #602: Set clock to Windows 10 workstation?

therentabrain, 2023-03-16:
I have disabled Internet access for my y291ga camera and am using it with the cloud off.   It can't reach an official time server.  But it should be fine reaching a Windows machine on my own LAN, right?  

I have tried putting in the machine's name and its IPv4, and setting the time zone properly, and it still thinks it's 2020.  (I understand; I still think it's 2020 sometimes.)  Is it possible to get the date/time from my own computer here?  I checked and my other Windows computers can get the time from it.

Thanks!!

roleoroleo, 2023-03-16:
You can set your own ntp server.
But I don't know if Windows provides a standard ntp service.

dieideeistgut, 2023-03-16:
Maybe in your case just using the Routers NTP works fine.

therentabrain, 2023-03-16:
In theory it does provide NTP service, but in practice I'm not sure.

I have luci on my router and was able to turn on the NTP server under System, Time Synchronization, and giving the router IP as my NTP server allowed the camera to work right away!

closed 2023-03-16 by therentabrain

therentabrain, 2023-03-16:
I found this link; this would probably work for anyone else who finds this thread:
https://support.industry.siemens.com/cs/document/22144502/