roleoroleo/yi-hack-Allwinner-v2 #605: Yi Home 1080p LOG file?

KayGundhardt, 2023-03-25:
Hi there,

I am using two YiHome 1080p cams actually with y291ga v0.2.6 (9.0.35.00_202111291330). Since ever they both work like a charm, but one of them gets unreachable after several days. Reboot helps - but only for a few days again.

I have to questions:

First:
How can I get a log out of the cam? 
Via SSH I can find 

```
log_first_login.tar.gz     
log_wifi_connected.tar.gz
log_login.tar.gz

```

 in `/tmp/sd/log` but cannot understand what i.ex.

```
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
```

means. Are there any other log files?

Second:
As a workaround: is it possible to write a daily automatic reboot script? Has anyone done this yet?

Any help would be really appreciated!

roleoroleo, 2023-03-25:
No, at the moment there are no logs.
I'm working on it for the next release.

KayGundhardt, 2023-03-25:
Great, thank you!

As for now, it seems to be the same issue as described [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/603): and the "swap-file-activation-workaround" works (for now ;)

closed 2023-07-24 by stale[bot]