roleoroleo/yi-hack-Allwinner-v2 #607: No web interface http://IP-CAM:8080

MrCykor, 2023-03-28:
Hello,

Can u help me, i hacked the camera, i see theese files:
[image]

i suppose if filename = **Factory.done** then camera is hacked, but i dont see web interface.

Any suggestions?

MrCykor, 2023-03-28:
@roleoroleo i know IP-CAM, nad skan it with nmap:
Only one port is open
PORT   STATE SERVICE    VERSION
23/tcp open  tcpwrapped

roleoroleo, 2023-03-28:
Ok, if just telnet is started, the main script is hacked but the script in the sd card is not executed.
Try with another sd.

MrCykor, 2023-03-29:
I switch card to smaller 4GB, format it using yi apk and still the same.
Capacity of cdcard is matter?

feeleep75, 2023-04-24:
I have the same problem but I noticed my BFUS camera have newer firmware than recommended: 12.2.0.2 - may this be a problem?

roleoroleo, 2023-04-26:
Could be.
Please, make a backup copy of the new firmware and share it with me.

romainbeaumont, 2023-04-27:
Hello,

I've the same issue : just 23/tcp port for telnet is open.. 
I've Dome Guard Camera.

feeleep75, 2023-04-27:
i just realised that for firmware 12.2.0.2 you should use this repo: https://github.com/roleoroleo/yi-hack-Allwinner - works for me

romainbeaumont, 2023-04-27:
Can you show me the tree structure of your sd card during the hack? Also, do you know if I can rehack it by simply pressing the reset button? I also noticed one thing: it doesn’t seem to connect properly to the internet when the SD card is inserted.

roleoroleo, 2023-04-30:
feelep75 is right.

MrCykor, 2023-05-04:
Hi,
I try with https://github.com/roleoroleo/yi-hack-Allwinner but for me it doesnt change enything :( 
[image]
No respond from http://192.168.0.45:8080

roleoroleo, 2023-05-04:
What the Yi firmware version of the cam?

MrCykor, 2023-05-04:
Yi firmware version 9.0.19.02

roleoroleo, 2023-05-04:
Your cam is a y21ga.
Try with the proper hack.

roleoroleo, 2023-05-05:
Clean the sd card and try again the hack with y21ga.

AlexNikMak, 2023-07-02:
Hello!

Seemse that I have a similar issue. After successful hacking for some reason, I cannot reach WEB UI on my YI1080p Home http://192.168.1.10:8080.
[image]
 According to YI APP my Yi firmware version 12.0.37.0_202211241030 and model IFUSY6557Y. I were installing [y211ga_0.2.7.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.7/y211ga_0.2.7.tgz)
I tried switching sd cards 3 times(different manufacturers and capacities) but it did not help.
Pinging my ip cam is successful. 

I would be grateful to receive some help

roleoroleo, 2023-07-02:
The default port is changed to 80.
Try it.

AlexNikMak, 2023-07-02:
I tried it right now but still i am unable to connect to WEB UI

closed 2023-10-30 by stale[bot]