roleoroleo/yi-hack-Allwinner-v2 #611: Cruise mode on Yi Dome Guard failures

morenod, 2023-04-01:
Firmware Version | 0.2.6
-- | --
Base Version | 12.1.47.2_202301121850
Model Suffix | r35gb
Serial Number | RFUSY73xxx (Yi Dome Guard)

When selecting cruise mode 360, camera tries to infinite rotate, but it cant. this camera rotates 360 but with no infinite loop. it should change direction when arriving the limit

roleoroleo, 2023-04-05:
What's the behavior when you activate cruise mode from the app?

morenod, 2023-04-05:
you mean the yi app? there is no cruise mode on it.

if enabled on the yi-hack page, camera rotates until the limit, and keeps trying on it each few seconds, so it hangs in a forever loop

roleoroleo, 2023-04-06:
So you don't have the cruise menu?
[image]
[image]

Probably your cam doesn't support it.

morenod, 2023-04-11:
I dont see that menu, only a drop down switch to select cruise mode:

[image]

If I set to 360, it fails. I can only use presets or off

roleoroleo, 2023-04-13:
I mean the menu in the app, not in the web interface.

morenod, 2023-04-13:
> I mean the menu in the app, not in the web interface.

which app?

roleoroleo, 2023-04-15:
"Yi Home" app for mobile.
Probably you don't use the app but inside it you can see the supported feature of your cam.

closed 2023-08-13 by stale[bot]