roleoroleo/yi-hack-Allwinner-v2 #614: Camera hangs after one day

bocob5, 2023-04-07:
System
Hostname	Mansarda
Firmware Version	0.2.6
Base Version	9.0.36.00_202011101611
Model Suffix	y211ga
Serial Number	IFUSY65M633D2O210114
Local Time	Fri Apr 7 17:06:44 CET 2023
Uptime	0h 23m 3s
Load Average	1.03 1.40 1.24
Free/Total Memory	38408/60912 KB
Free SD Space	100%

After the install of the 0.2.6 version Yi Camera hangs after approximately one day. There is no ping or SSH available. Issue looks like some type of kernel crash. I also tried to solder wires on console pins, but somehow I do not get anything on the console. Minicom on linux was used. So, unfortunately I can't provide any diagnostic information that might help with the investigation. 
  
Yi Cam running  Yi firmware base Version 11.1.35.3 and Yi Hack version 0.2.5 does not show this problem. 
My Yi Cams where running OK on version 0.2.5. So it looks this is version 0.2.6 specific problem. 

Let me know if there is any further information I can provide in order to find root cause of this issue. 
For the time being I can live with this problem, since I added ESP32 power switch to the cameras and it is possible to remotely power On/Off the cameras.
 
Thanks

bocob5, 2023-04-14:
Thank you for the suggestion. 
I've already enabled swap on one camera, just to be sure I enabled it on another one, but results are the same. Cameras still hangs.

roleoroleo, 2023-04-15:
Reading your config file, swap seems to be disabled:

> SWAP_FILE=no

bocob5, 2023-04-18:
I have 4 cameras and swap was enabled on another camera, not the one config
was used in this issue report. But behavior is the same.
If you might find some time to look into this issue, what diagnostic info
could be interesting for you?
And thank you for this great solution you provided to the community in this
project.

roleoroleo, 2023-04-18:
When the cam reboots the reason could be:
1 - Not original power adapter or cable (it may seem silly but i have seen it happen many times).
2 - Power suppy problem.
3 - Memory problem (a process requires a lot of memory, the kernel starts to kill processes and the watchdog reboots the system).

In the last case you could try to rename the watchdog process wd_rtsp.sh and check what happens.

closed 2023-08-16 by stale[bot]