roleoroleo/yi-hack-Allwinner-v2 #618: Mqtt event motion_files empty files

iupsilon, 2023-04-22:
Hi,
I installed yi-hack-Allwinner-v2 on my Yi Home Camera

Firmware Version: 0.2.6
Base Version: 12.0.35.8_202211241029
Model Suffix: y291ga
Serial Number: RFUSY6217Z0853221206

I enabled MQTT and Motion Detection (Camera Settings)

Using a MQTT client I subscribed to all topics of "webcam"
I see :

webcam/stat/camera/save_video_on_motion yes
webcam/stat/camera/motion_detection yes
webcam/stat/camera/sensitivity medium
webcam/stat/camera/ai_human_detection no
webcam/stat/camera/face_detection no
webcam/stat/camera/motion_tracking no
webcam/stat/camera/sound_detection yes
webcam/stat/camera/sound_sensitivit 80

After a motion is detected I receive a message:

Topic: webcam/motion_files
{
"start":"2023-04-20T11:30:59+02:00",
"end":"2023-04-20T11:31:59+02:00",
"files":[ ]
}

"files" is always empty.

But I see that /tmp/sd/record/ contains the captured video.

I have several Yi Home cameras with different yi-hack versions and the event motion_files always contains the name of recorded files.
Many thanks!

Cheers.

roleoroleo, 2023-04-26:
Is timezone properly set?
Both in the app and in the web page?

iupsilon, 2023-04-26:
Hi,
yes, timezone is set:
CET-1CEST,M3.5.0,M10.5.0/3
and corresponds to Europe/Rome

Recorded videos are correctly created in sd , I see lots of them in /tmp/sd/record
for example:
/tmp/sd/record/2023Y04M26D19H/E200M00S60.mp4

Date and time is coherent with the recording.
Is there a way to enable mqtt event logging?

Many thanks for you work.

roleoroleo, 2023-04-30:
You could try to kill mqttv4 and restart it manually.
You should read the log in the standard output.

I'm pretty sure that it's a timezone problem but I don't understand why.
What's the date/time of the recorded file? It matches your local time or GMT time?

iupsilon, 2023-04-30:
Hi,
system datetime is GMT
date -> Sun Apr 30 21:53:52 GMT 2023

Record files are create in GMT format

I see process mqttv4 is launched with parameter -t local
mqttv4 -t local

In script: script/system.sh
I see:

if [ "$HV" == "11" ] || [ "$HV" == "12" ]; then
    mqttv4 -t local &                                                                                                  
else                                                                     
    mqttv4 &                                                             
fi  

cat /home/homever -> 12.0.35.8_202211241029
HV=${HOMEVER:0:2} -> 12

If I lauch mqttv4 without -t local everything works fine and
Topic: webcam/motion_files
contains the recorded file as expected.

Cheers.

roleoroleo, 2023-05-01:
Strange behavior.
I will add an exception for your model.

roleoroleo, 2023-05-01:
Cloud is enabled or disabled?

iupsilon, 2023-05-01:
I tried both enabled and disabled but the result is the same.
The local time is UTC and recordings are saved with filename containing UTC reference.

ema987, 2023-08-02:
Hi all,

exactly the same issue seems to happen to me when using:

```
Firmware Version: 0.2.9
Base Version: 12.0.37.0_202211241030
Model Suffix: y211ga
Serial Number: QFUS*
webcam/stat/camera/switch_on=yes
webcam/stat/camera/save_video_on_motion=yes
webcam/stat/camera/motion_detection=yes
webcam/stat/camera/sensitivity=low
webcam/stat/camera/ai_human_detection=no
webcam/stat/camera/ai_vehicle_detection=no
webcam/stat/camera/ai_animal_detection=no
webcam/stat/camera/face_detection=no
webcam/stat/camera/motion_tracking=no
webcam/stat/camera/sound_detection=yes
webcam/stat/camera/sound_sensitivity=80
```

Applying the changes suggested by @iupsilon , so removing `-t local` from `script/system.sh` fixed the issue and files started appearing in `camera/motion_files`.

Before upgrading to 0.2.9/12 the same webcam was on 0.2.6/9 and everything was working correctly, the issue appeared to me after I upgraded to 0.2.9/12 today.

If you need any help debugging the issue let me know, hopefully I can help. Thank you for the hard work!

roleoroleo, 2023-08-21:
I will add another exception for y211ga.

closed 2023-12-19 by stale[bot]

darkfuncat, 2024-01-09:
I have the same case with the new Yicam 2k pro (y623).
I understand that you have to configure the model manulaly to fix this issue ?

darkfuncat, 2024-01-11:
I tried, it works now when I remove "-t local" (model y623) thanks