roleoroleo/yi-hack-Allwinner-v2 #620: Can't access HTTP using newly created login/password.

ikhimichenko, 2023-04-23:
Hi guys,
First of all thank you a lot for your efforts on this project!
So I've a BFUS yi 1080p home camera, flashed it and just added username/password fields (default were empty):
[image]
Then rebooted camera.
Now I can't access camera by HTTP , 'cause login/password are incorrect.
Only re-flash helps.
Is there a way to set username/password ?
Thank you.

ikhimichenko, 2023-04-26:
My apologies, I have to use '#' sign.
Sorry and thanks for your efforts.

closed 2023-04-26 by ikhimichenko

ikhimichenko, 2023-04-26:
@roleoroleo last question here:
I set login/password. But how can I reach this camera by ftp? I tried to use mentioned login/password but ftp server says "530 Login authentication failed"

roleoroleo, 2023-04-29:
ftp login is with root user.