roleoroleo/yi-hack-Allwinner-v2 #622: Time in events

JozefLu, 2023-05-02:
Hi,
yi-hack-Allwinner-v2 on my Yi Home Camera:
Firmware Version	0.2.6
Base Version	12.0.35.8_202211241029
Model Suffix	y291ga
Serial Number	RFUSY62xxxxxx
Motion detection is ON.
Time Zone in hack is: CET-1CEST,M3.5.0,M10.5.0/3
and is the same in app
But the time in Events is shifted 2h and I think is GMT, not mine.

roleoroleo, 2023-05-04:
What do you mean with "But the time in Events is shifted 2h and I think is GMT, not mine."?
Where do you see GMT?

JozefLu, 2023-05-06:
O, sorry, I did not expleined it clearily.
Below is an example - the time is 10:00:
[image]

Inside is the same time 10:xx:
[image]

But in the reality is 12:xx (the right time is only shown as a timestamp in the recording file):
[image]

roleoroleo, 2023-05-14:
Please, try to apply this patch:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/8bb5c60e0c18a4bcca1d61d00b703b368b455c43

thecount2a, 2023-06-13:
I think the problem here is with devices with built-in device OSDs don't get a TZ offset set at all.  Since I have the t211ga firmware, my OSD is also wrong unless I add this line to startup.sh:

set_tz_offset -c tz_offset -v -14400

The system.sh script runs this command but it does nothing for my built-in OSD:

set_tz_offset -c tz_offset_osd -m y211ga -f 12 -v -14400

As a workaround @JozefLu you can add a similar line for your timezone offset to a file in your root directory called "startup.sh"

@roleoroleo I think this bug has something to do with negative timezone offsets not working correctly (maybe the negative sign is being dropped) because I can set positive timezone offsets just fine but it treats negative offsets as positive offsets.  For my -4 hour timezone offset, it adds 4 hours instead of subtracting 4 hours.  EST-4 gives me the same OSD value as EST4, but the system time is different for those two settings.  I can set my timezone to EST20 and get the correct time but then the date value is 1 day ahead!

michal0125, 2023-09-08:
I have the same issue. 

Firmware Version | 0.2.9
Base Version | 12.0.37.0_202211241030
Model Suffix | y211ga
Serial Number | IFUSY65xxxxxxxxxxx

roleoroleo, 2023-09-09:
Please, recap the problem.
I have released a couple of versions since this issue was opened.

pafade89, 2023-09-18:
Hi @roleoroleo,

I have the same issue with my IFUS Home Cam. The Local Time on the landing page and the watermark time in the video are correct. The event time on the event page is 2 hours too early.

Same like @JozefLu 

Cheers P

roleoroleo, 2023-09-24:
Do you mean the 1st column of the page "Date & Time"?

Please, give me some details:
- suffix
- fw version
- hack version
- time zone configured

pafade89, 2023-09-24:
Hi,

Timezone is:

> CET-1CEST,M3.5.0,M10.5.0/3

Here are the Cam Infos

[image]

And here the problem:

[image]

The Event URL is also wrong:

http://IP:Port/?page=eventsfile&dirname=2023Y09M24D16H

roleoroleo, 2023-09-26:
What's the time configuration in the app?

pafade89, 2023-09-26:
[image]

roleoroleo, 2023-09-30:
Sorry, I need time to fix this issue.

pafade89, 2023-12-29:
Unstale

roleoroleo, 2024-01-05:
I'm not able to autodetect the correct time for all models.
So I added a configuration option.
Please, try this beta:
[h30ga_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13841245/h30ga_0.3.0.tar.gz)

roleoroleo, 2024-01-05:
yi-hakc/www and yi-hack/script folders are enough.

And a reboot before to configure it.

pafade89, 2024-01-08:
After switching to 'local time' in the drop-down menu, saving, and rebooting, the menu reverts to 'auto detect'.

It also has no effect on the timestamp.

roleoroleo, 2024-01-08:
> I tried it today, set it to GMT, but the folder name still doesn't match, so the link isn't right (see screenshot).

This is normal.
The name of the folder is GMT.
But the column "Date & Time" must be ok, aligned with the local time.

roleoroleo, 2024-01-11:
The time you see in the app should be the same as the time in the "Date & Time" column (or "Time" column in the 2nd web page).
The "Directory name" column, however, is different: it's in GMT time because the folder is created by a Yi process that works in GMT.
So, there is a translation between the folder and the event time.

If you are in UTC+1 timezone, in this moment you should see a 1 hour offset between them (for your cam model).
In this post https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/622#issuecomment-1881440682 the behavoir seems correct.
In this post it's not ok https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/622#issuecomment-1883129202.

For your specific model you should set "GMT". The meaning of the switch is: "what time your cam is using to save video files?"
If the folders in /tmp/sd/records are saved with GMT time, set "GMT".
If the folders in /tmp/sd/records are saved with local time, set "local time".

pafade89, 2024-01-11:
@roleoroleo 
Thank you for your explanation. However, the question that now arises for me is why my other three Yi Outdoor Cameras display everything perfectly. Two of them have the Yi Hack 5, and the other two have the Yi Hack Allwinner v2 from you, out of my total of 4 cameras.

Yi Hack Allwinner v2:

[image]
[image]

Yi Hack v5:

[image]
[image]

[image]
[image]

I hope that helps, maybe.

pafade89, 2024-01-15:
I thank you for the implementation, and I assume we can close the issue, right?

roleoroleo, 2024-01-16:
In the next release, I completely changed this behavior.
But we will open a new issue, if necessary.

pafade89, 2024-01-18:
That sounds great =) I'll wait for further updates.

#close this issue

closed 2024-01-18 by roleoroleo