roleoroleo/yi-hack-Allwinner-v2 #623: ftppush.sh does not detect already running script

maciekn, 2023-05-03:
Environment:
Firmware Version | 0.2.6
Base Version | 9.0.19.02_202007141512
Model Suffix | y21ga

It looks `ftppush.sh` detection of already running script fails causing it to duplicate. Caught this while looking for potential reasons for spontaneous reboots.
Example 'ps check':
`
# ps | grep ftppush.sh | grep /bin/sh
 2020 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
18023 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
27595 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
`
while looking at the code there should be only one instance running. Debug in progress :)

maciekn, 2023-05-03:
I've modified the script to persist some extra information in the bootstrap of 'cron' code branch. Added just below running process detection:
```
	logPersist "Running: $RUNNING"
	logPersist "ps | grep $SCRIPT_FULLFN"
	logPersist `ps | grep $SCRIPT_FULLFN`
```

Response (reformatted for better readability):
```
2023-05-03 [15-16-40] Running: 0
2023-05-03 [15-16-40] ps | grep ftppush.sh
2023-05-03 [15-16-41] 
5157 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
5403 root 1976 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
5404 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
19232 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
19326 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
19418 root 1004 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
19444 root 1004 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
19447 root 980 S grep ftppush.sh 
29221 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
29692 root 1976 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush 
29693 root 1384 S {ftppush.sh} /bin/ash /tmp/sd/yi-hack/script/ftppush
```

While `ps | grep ftppush.sh | grep /bin/sh` run few minutes later yields:
```
 2020 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
18023 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
27595 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.sh cron
```

roleoroleo, 2023-05-04:
We could create a semaphore file in /tmp and check if it exists when the script starts.

maciekn, 2023-05-06:
Semaphore I think would do the trick.

However, a bit by trial-and-error I think I made the script working. Long story short, I did it by replacing:
```
if [ "${1}" = "cron" ]; then
	RUNNING=$(ps ww | grep $SCRIPT_FULLFN | grep -v grep | grep /bin/sh | awk 'END { print NR }')
``` 
to:
```
if [ "${1}" = "cron" ]; then
        RUNNING_SRC=$(ps ww| grep $SCRIPT_FULLFN )
        RUNNING_SRD=$(echo "$RUNNING_SRC" | grep -v grep)
        RUNNING_SRE=$(echo "$RUNNING_SRD" | grep /bin/sh)
        RUNNING=$(echo "$RUNNING_SRE" | awk 'END { print NR }')
```

so it is either `ps ww` which does the trick or separation into separate lines. Some not anticipated busybox behavior?

After doing this change cam seems to be running stable for more than 2 days and counting:
Uptime | 2d 9h 22m 30s
Load Average | 2.54 2.48 2.51
Free/Total Memory | 31552/60912 KB

roleoroleo, 2023-05-11:
I will add this change to the next release.
Thank you.

maciekn, 2023-05-11:
@roleoroleo I'd be more than happy to debug it a bit more in my spare time. I still feel a bit anxious not understanding why it is behaving like this. 
Any tips on how to add something to cron? I've noticed that the unexpected behavior happens only when script is fired from cron and not manually.

roleoroleo, 2023-05-12:
For testing purpose, you can edit the file /var/spool/cron/crontabs/root

maciekn, 2023-05-14:
I think I got do the root of the problem. Looking at grepping part:
```
SCRIPT_FULLFN="ftppush.sh"
(...)
ps ww | grep $SCRIPT_FULLFN
```
the `ps` output got truncated to meet the line character limit:
```
(...)
 6310 root       984 S    -ash
11032 root         0 SW   [kworker/0:0]
12310 root         0 SW   [kworker/u2:0]
13175 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.
```
Hence line has been truncated, it did not match the grep predicate. Adding `ps ww` fixes the issue.
PR linked above

roleoroleo, 2023-05-16:
Thank you for your work.

closed 2023-09-13 by stale[bot]