roleoroleo/yi-hack-Allwinner-v2 #628: always night vision?

codekm92, 2023-05-09:
Hello,
is it possible to enable a always night vision mode? i record videos at night with my cam but in the room 
is a TV and now the cam always hops between night vision and back. i want only nightvision.

Maybe there is a settings i can set?

roleoroleo, 2023-05-14:
As far as I know, it's not possible.

ch3funio, 2023-07-25:
Hi! I'd also like to have full control over the night vision mode/remove the IR filter from the lens, it causes the typical annoying "click" sound or some configuration of the light sensor so it doesn't switch night mode so often

Gifford47, 2023-10-28:
On current firmware there's a switch to enable/disable ir led. But in my case it switches ir led on/off as before depending on the brightness.
@roleoroleo Is it a bug or depends it on the hardware?
My y21ga is on 0.30.

roleoroleo, 2023-11-03:
If I remember correctly, the meaning of the switch should be:
- enable: switch on/off automatically depending on the brightness
- disable: completely disabled

If you want another behavior (e.g. enable day and night) it's not possible.

closed 2024-03-03 by stale[bot]