roleoroleo/yi-hack-Allwinner-v2 #639: Error changing WIFI settings but saved

fernandolcardoso, 2023-05-19:
When editing the WIFI settings, an error is displayed, but when restarting the camera, the new settings are accepted.

[image]

[image]

roleoroleo, 2023-06-03:
Here is the code:
```
    read -r POST_DATA
    rm -f /tmp/configure_wifi.cfg

    # Validate json
    VALID=$(echo "$POST_DATA" | jq -e . >/dev/null 2>&1; echo $?)
    if [ "$VALID" != "0" ]; then
        printf "Content-type: application/json\r\n\r\n"
        printf "{\n"
        printf "\"%s\":\"%s\"\\n" "error" "true"
        printf "}"
    fi
    KEYS=$(echo "$POST_DATA" | jq keys_unsorted[])
    for KEY in $KEYS; do
        KEY=$(echo $KEY | removedoublequotes)
        VALUE=$(echo "$POST_DATA" | jq -r .$KEY)
        if [ $KEY == "WIFI_ESSID" ]; then
            KEY="wifi_ssid"
            echo "$KEY=$VALUE" >> /tmp/configure_wifi.cfg
        elif [ $KEY == "WIFI_PASSWORD" ]; then
            PWD=$VALUE
            KEY="wifi_psk"
            echo "$KEY=$VALUE" >> /tmp/configure_wifi.cfg
        elif [ $KEY == "WIFI_PASSWORD2" ]; then
            PWD2=$VALUE
        fi
    done

    if [ "$PWD" == "$PWD2" ]; then
        $YI_HACK_PREFIX/script/configure_wifi.sh
        sleep 1
        rm -f /tmp/configure_wifi.cfg

        printf "Content-type: application/json\r\n\r\n"
        printf "{\n"
        printf "\"%s\":\"%s\"\\n" "error" "false"
        printf "}"
    else
        rm -f /tmp/configure_wifi.cfg

        printf "Content-type: application/json\r\n\r\n"
        printf "{\n"
        printf "\"%s\":\"%s\"\\n" "error" "true"
        printf "}"
    fi
```
It exits with error only for 2 reasons:
- Error parsing the json content from the browser
- Password 1 and password 2 differ

Which special char are you using?

fernandolcardoso, 2023-06-03:
Hello  @roleoroleo, I believe I was not clear in my statement.

I can successfully change the WIFI password, the issue is that even if I succeed, the error is returned.

I would just like to check why it is returning error.

roleoroleo, 2023-06-04:
I understand.
At the end of this if statement https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/44daeff5fb6fcbb7667ae9c11f0074861dcbdbd2/src/www/httpd/cgi-bin/wifi.sh#L53
the script should exit but there is an error.
So the script sets error=true but it continues to write your credentials.

I'm trying to repeat this behavior with my cam but I'm not able: it completes suyccessfully.

closed 2023-10-02 by stale[bot]