roleoroleo/yi-hack-Allwinner-v2 #646: 'human_detection' permanently active in HA

cjgarciao, 2023-06-14:
Hello,
I have a problem with human detection, I think after upgrading to 0.2.7, with a couple of Dome Guard cameras in HA.
After a while running, the 'human_detection' sensor remains, permanently, active.
I have used 'mosquitto_sub' to see the 'motion_detection' topic and I have observed that the 'human' message is sent when there is someone, but when there is no more person, the 'motion_stop' message is not sent (neither 'human', until someone appears again, that 'human' is sent again correctly).
Should I change something in the configuration?
Best regards,
Carlos.

roleoroleo, 2023-06-14:
Strange behavior. There are no changes in this part of code.
What about a normal motion detection? Does it work?

roleoroleo, 2023-06-14:
Which cam do you have?
Suffix? fw version?

cjgarciao, 2023-08-21:
I keep having the same problem on several cameras, always with firmware version 12 or 11, with version 9 it doesn't happen anymore.
Specifically, it fails me with the following cameras and firmware (the hack version is 0.2.9 in all cases):

r35gb - fw 12.1.47.2_202301121850
h30ga - fw 11.1.26.1_202301191526
y291ga - fw 12.0.35.8_202211241029

I have another Yi outdoor (h30ga) with the old firmware (9.0.26.07_202104231123), with the 0.2.9 hack and it works correctly.

There are times that it works correctly for hours or even days after rebooting the camera. I have deleted the settings on several of them and re-entered them manually, but the behaviour is the same.
I imagine the problem might be in the MQTT part or some more internal message in the camera that doesn't release the detection. Any ideas I can try?

Carlos.

roleoroleo, 2023-08-24:
Do you switch on and off the cam frequently (for example with a script)?
Is it possible that the cam doesn't catch the motion stop because is switched off?

cjgarciao, 2023-08-28:
No, I never switch off the camera.

roleoroleo, 2023-09-01:
I'm not able to replicate your problem with my cam.
When I enable human detection, I receive properly the stop message.
Could you test if the app works properly?
I would like to understand if the problem is the mqtt part.

fnordson, 2023-09-19:
I've had the same issue with Sound Detection, but not in HA. On the camera's web interface. 

Enabled it at first just to play around. It showed "disabled" when I went to the appropriate settings afterward. 
So I enabled it again and disabled it.  
Now it's off. 

Won't touch it again now, but I assume it's the same issue with "ONVIF watermark" since I can't turn it off even when it shows to be off and since I'm not sure how "Face Detection" works, I won't touch it even though it's on now. 

I have a Dome Guard too:

Firmware Version | 0.2.9
Base Version | 9.0.46.00_202110081405
Model Suffix | r35gb

closed 2024-01-18 by stale[bot]