roleoroleo/yi-hack-Allwinner-v2 #648: Recording 3 minutes of video in the time period from 15:00 to 15:30

antonsrc, 2023-06-14:
Hello!
I'm not Linux guy, but is it possible somehow to record a small video every day through the sh script and Crontab? For example, according to the schedule, record once a day time period from 15:00 to 15:30 a 3-minute video and save it to SD-card?

_Camera spec:_
Yi Outdoor O30
YHS.3020
**h30ga**

Thanks!
Anton

roleoroleo, 2023-06-15:
I don't remember if it's possible to make a video bigger than 1 minute.
But you can try.
Check the wiki for the web service command:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#record
Or in shell:
`ipc_cmd -S 180`

antonsrc, 2023-06-15:
Those. create a script posted for example here?
`* 15 * * * /tmp/sd/yi-hack/my_script.sh`

Well, if I only write one minute, then should I write this line in the script?
`ipc_cmd -S 60`

roleoroleo, 2023-06-15:
Yes, you can create a script and call it from cron.
And yes, the last parameter is the time in seconds.

antonsrc, 2023-06-15:
Records, but unfortunately less than 1 second ...
Do you have any idea how to increase the recording time to 60 seconds?

antonsrc, 2023-06-26:
Thank you! It works!

closed 2023-06-26 by antonsrc