roleoroleo/yi-hack-Allwinner-v2 #650: Can you make the same for Escam pt200?

alper061, 2023-06-17:
Can you make the same for Escam pt200?

This is the camera:
http://www.escam.cn/product/223-en.html
https://www.banggood.com/ESCAM-PT200-2MP-1080P-5G-Dome-WIFI-IP-Camera-Mobile-Tracking-Coud-Storage-Bidirectional-Voice-Night-Vision-Home-Security-CCTV-Monitor-p-1948164.html?cur_warehouse=CN&p=6B232338637912016021&utm_campaign=22331227_1874&utm_content=3547&ID=47757

roleoroleo, 2023-06-18:
Try to make a backup of the cam.
If it works, maybe I can try to support this cam.

alper061, 2023-06-27:
Okay. I'll try it and upload if it works.

alper061, 2023-06-28:
I tryed to upload the backup .sh files, but camera starts and after 50 sec it reboots and it repeats. So if the card is in, the camera will rebooting constantly.

alper061, 2023-06-28:
The reboot happens if i copy the backup files to the sd. If its blank, the camera starts normal

roleoroleo, 2023-06-28:
Maybe a problem with this card?
Please try another sd or try to reformat it.

alper061, 2023-06-28:
It is a brand new card. If its blank, the cam put some folders and files when it starts. (recordings folder, a .db file). I tryed to reformat. When i put the backup files to the card, camera bein to restarting.

alper061, 2023-06-28:
So the card is probably working.

alper061, 2023-07-03:
tryed with 2 more cards and no luck :(

roleoroleo, 2023-07-03:
Even if the cam restarts, is the backup created?

alper061, 2023-07-03:
No, unfortunately. :(

roleoroleo, 2023-07-03:
Which OS are you using?
Which unzip program are you using?
Make sure that the sh file is a linux file and not a windows file.

alper061, 2023-07-03:
I'm using windows and 7zip.

alper061, 2023-07-03:
I tryed to delete the dd lines from config.sh. The change is faster restarts :)

With original config.sh:
- Start cam
- Wait 20-30 seconds
- Restart
- Repeat

With modifyed config.sh (deleted the dd lines):
- Start cam
- Wait 5 seconds
- Restart
- Repeat

alper061, 2023-07-30:
Its not solwed. What needs to do to make forward with this?

roleoroleo, 2023-08-18:
> I tryed to delete the dd lines from config.sh. The change is faster restarts :)
> 
> With original config.sh:
> 
>     * Start cam
> 
>     * Wait 20-30 seconds
> 
>     * Restart
> 
>     * Repeat
> 
> 
> With modifyed config.sh (deleted the dd lines):
> 
>     * Start cam
> 
>     * Wait 5 seconds
> 
>     * Restart
> 
>     * Repeat

This means that dd commands are executed.
But the output file are not saved inside the sd card and I don't know why.
Probably the fw is different and the mount point is not /tmp/sd
I don't know how to fix this problem.

closed 2023-12-16 by stale[bot]