roleoroleo/yi-hack-Allwinner-v2 #651: PCM sound RTSP streaming not working, in 0.2.7, y291ga 9.0.35

hj2650, 2023-06-19:
Hi,

PCM audio streaming used to work in 0.2.6, but in 0.2.7, there is no sound output.

ACM audio streaming is working, however.

hj2650, 2023-06-19:
And after changing the audio streaming option to ACM, the stream is very unstable -- a frame arrives in 2 mins.

roleoroleo, 2023-06-19:
This is a very insidious and recurring problem.
When I enable -O2 optimization flag in gcc, the resulting binary doesn't work properly.
I have to set -O0.
I will release a new versione ASAP.
Meanwhile use this:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11787372/rRTSPServer.gz)

hj2650, 2023-06-20:
> This is a very insidious and recurring problem. When I enable -O2 optimization flag in gcc, the resulting binary doesn't work properly. I have to set -O0. I will release a new versione ASAP. [...]

Thank you for the fast reply. Could you please link me a document/wiki page or just briefly explain how to apply the rRTSPServer.gz package file on a camera?

hj2650, 2023-06-21:
Now It works flawlessly. Thank you.

closed 2023-10-28 by stale[bot]