roleoroleo/yi-hack-Allwinner-v2 #655: Hardware donation

roleoroleo, 2023-06-23:
I'm having trouble maintaining support for all cams because I have very few models available. Especially older ones that are no longer available for purchase.
If you have cameras that you don't use anymore (maybe broken) and you want to get rid of them, please send to me.

Giuserver, 2023-06-30:
ho 2 telecamere che posso donarti. (ma entrambi senza infrarossi e obiettivo, solo perché le ho riutilizzate in altre telecamere, ma puoi adattarci quelle che hai) Il problema delle 2 telecamere:
La prima, il led rimane giallo fisso (credo sia un problema di voltaggio) la seconda si riavvia quando si connette al wifi. Posso interessarti? 
They're y21ga

roleoroleo, 2023-07-01:
Thank you for your offer but I own 2 of that model.
So I don't need it at the moment.

alper061, 2023-08-04:
I can send you some cameras. Please write an e-mail.

roleoroleo, 2023-08-19:
Write me here roleo@iperbole.bologna.it

bocob5, 2024-05-14:
I can send you camera with the SN range IFUS if needed.

roleoroleo, 2024-05-15:
Which model suffix?

bocob5, 2024-05-20:
Sorry, I was not specific enough. Hope this is info you requested:

Product Name: YI 1080p Home Camera
Model YYS.2016

roleoroleo, 2024-05-20:
Yes, I need it.
Send me a mail for details (the mail is in the paypal link).

roleoroleo, 2024-05-25:
I sent you an email.
Remove your address above if you want.
Thank you.

roleoroleo, 2024-10-04:
Now, I would need a y623...

frosty-geek, 2024-10-08:
@roleoroleo I have a y623 which I can donate, would you rather want a money donation or the actual device? (I think shipping it would be roughly the same)

roleoroleo, 2024-10-11:
I would prefer the cam.
At the moment the y623 is giving me a lot of headaches.
Thank you.

roleoroleo, 2025-01-11:
Up

closed 2025-05-14 by github-actions[bot]