roleoroleo/yi-hack-Allwinner-v2 #659: problem with mqtt remote configuration

peterkowalewski, 2023-06-26:
I have got some issue with mqtt. I have the camera h30ga. After installing 0.2.7 version over the original yi 9.0.26 I was not able to configure the topic for remote configuration.
The filed in the web UI was gray out. Even after the configuration of "MQTT Camera Setting Advertise", editing was not possible. I added to the mqttv4.config „MQTT_PREFIX_CONFIG=yicam-cfg“  rebooted the camera and was able to publish to camera. But what I don’t understand publishing to yicam-config was not working. I have to publish to MQTT_PREFIX in this case default "yicam".  Why? After the reboot the remote configuration topic is editable in the web UI. Was this an expected behavior?
Piotr

roleoroleo, 2023-06-26:
Sorry, I temporarily removed this feature in this commit.
In this moment it's not working.
I will restore it in the next release.

peterkowalewski, 2023-06-27:
This explains. Thanks for the answer.

closed 2023-10-28 by stale[bot]