roleoroleo/yi-hack-Allwinner-v2 #664: Status of the camera is not correct

ttarzan01, 2023-07-08:
I use firmware 0.2.7 with model y291ga. Unfortunately, I can no longer read the status with IP:8080/cgi-bin/get_configs.sh?conf=camera. It always says "SWITCH_ON": "yes" even if the camera is turned off. What is the reason, what can I do?

roleoroleo, 2023-07-10:
This is not the correct behavior.
If you switch off the cam from camera_settings web page and click "Save Configuration", the cam should reply "SWITCH_ON":"no"
Please clean the cache of the browser and check again.

ttarzan01, 2023-07-10:
Sorry, this approach shows no effect. Everytime ist "SWITCH_ON":"yes".

roleoroleo, 2023-07-10:
Please, use the debug of the browser and try to capture the network request.
F12 with Chrome, network tab.

ttarzan01, 2023-07-12:
Ok. Camera is off, but camera-setting-page show "on".

[image]

This ist the network request:

[image]

roleoroleo, 2023-07-13:
Ok.
Now press Save button and capture the request and the response:

[image]

[image]

thoemmes86, 2023-10-15:
I face the same issue.
Looks like after some time 1/2 day camera switched off, it is not longer possible to turn on/off the camera via the hack web interface or the applicable request.

Workaround is to use the yi app to get it again turned on.

captaincanada, 2023-11-10:
could someone figure out whats the problem?
I have the same issue...

Few times ago i got the correct values within the response of  
"... cgi-bin/get_configs.sh?conf=camera"

but now theres nothing changing after switch something 
neither from the app, nor from Webpage

i used to resolve the json file for a state sensor in Mediola AIO Remote and wanted to use these 
at Home Assistant now.

Maybe this is also the Reason why the HASS Integration does not work for me...?!

Thx to All

roleoroleo, 2023-11-11:
Please explain me better.
The issue refers to an old version of the hack.

captaincanada, 2023-11-11:
Hi, thank you for taking care! 

the states of the camera are not correct when I check them with the "conf=camera" request.

this is always the answer, no matter what I use to change:

{
"SWITCH_ON":"yes",
"SAVE_VIDEO_ON_MOTION":"yes",
"SENSITIVITY":"low",
"AI_HUMAN_DETECTION":"no",
"SOUND_DETECTION":"no",
"SOUND_SENSITIVITY":"80",
"LED":"no",
"ROTATE":"no",
"IR":"yes",
"HOMEVER":"9.0.19.02_202007141512",
"NULL":"NULL"
}

roleoroleo, 2023-11-12:
Maybe you need a refresh of the cache.
If the js script is not synchronized the "Save" function doesn't work.
Please clean the cache and try again.

ttarzan01, 2023-11-13:
> cache is empty. when I save changed settings and refresh the page, it doesn't keep the changes...
> 
>  Screen_Recording_20231113_112526_Samsung.Internet.mp4

Same here.

roleoroleo, 2023-11-13:
Please, open the debug console in the browser and click "Save configuration".
Then check the response:
[image]

ttarzan01, 2023-11-13:
> Please, open the debug console in the browser and click "Save configuration".
> Then check the response:

I do not receive such a response.

It does not matter whether the camera is on or off, it is always displayed as on. It doesn't matter whether I call 'cgi-bin/get_configs.sh?conf=camera' or 'index.html?page=camera_settings'. 

I noticed another point: If I set the rotation to 'on' under 'index.html?page=camera_settings' so that the image is rotated, this setting is not displayed either. It always shows 'off' when I have saved the settings. Same with 'Status led'.

[image]

roleoroleo, 2023-11-13:
> I do not receive such a response.

Please share the request.

ttarzan01, 2023-11-13:
Ok, I cleared the browser cache, then called 'index.html?page=camera_settings'. The camera is displayed as 'on' even though it is off.
[image]
In the console I only see the following error:

[image]
Nothing more.

captaincanada, 2023-11-13:
[image]

ttarzan01, 2023-11-13:
After refresh:

[image]

But camera is off.

captaincanada, 2023-11-13:
it sends switch_on= on and switch_on=yes and the camera is switching on and of too, 
but eyerytime i ask for the state it says 

"SWITCH_ON":"yes"

same for browser and postman

roleoroleo, 2023-11-14:
Check if the file /tmp/sd/yi-hack/etc/camera.conf changes when you switch on/off.

ttarzan01, 2023-11-14:
> Check if the file /tmp/sd/yi-hack/etc/camera.conf changes when you switch on/off.

Do I have to take the SD out of the camera? Or can I access it in another way?

roleoroleo, 2023-11-14:
Use ssh.

captaincanada, 2023-11-14:
I did it with ftp and the files show exactly the same...

https://github.com/roleoroleo/yi-hack-Allwinner-v2/assets/2896314/bb29e0dd-8b5a-4a5e-bb36-4e207f959c28

captaincanada, 2023-11-14:
when I manually edit the camera.conf file to switch_on=no... it shows the right state on the web server.

But the effect is the same... it doesn't change the settings set from the web server... it's falling back to state "no"

roleoroleo, 2023-11-14:
I don't understand. Probably 2 different problems.
The status it' not correct but the question is: does the cam change the internal setting when you click "Save"?
Does the file camera.conf change?
So, the problem is the save method or just the get method?

captaincanada, 2023-11-14:
the camera.conf does not change.

1.switch web server to "on" 
> camera turns on

-web server button changes to on
-camera.conf doesn't change
-http request doesn't change (thought this would get the data from the camera.conf file)

2.switch web server to "off" 
(whiteout refreshing the web page, cause the button would change by it self after a refresh...)
> camera turns off

-web server button changes to off
-camera.conf doesn't change
-http request doesn't change

the web server button keeps its state and the camera.conf doesn't change. whatever I do
so I guess  the problem is the save method of the camera.conf

ttarzan01, 2023-11-16:
I have the same behavior of camera.conf on the SD like @captaincanada . No matter what I do, camera.conf doesn't change. Although the camera goes on or off as desired.

roleoroleo, 2023-11-18:
If the camera switch on/off but doesn't save, it's probably a problem related to mqttv4.

ttarzan01, 2023-11-18:
It says

root@xxx:~# ps | grep mqttv4
 1403 root       792 S    mqttv4
11639 root       980 S    grep mqttv4

roleoroleo, 2023-11-18:
mqttv4 is running.
Now try to kill and restart it manually.
```
# killall mqttv4
# mqttv4
```
and click the save button in the web gui.
You should see a log with:
```
...
CALLBACK COMMAND
Updating file "/tmp/sd/yi-hack/etc/camera.conf", parameter "switch_on" with value "no"
Client yi-cam-a2 sending PUBLISH (d0, q1, r1, m12, 'home-assistant/spare/stat/camera/switch_on', ... (2 bytes))
...
```

ttarzan01, 2023-11-18:
Only see this:

[image]

[image]

captaincanada, 2023-11-18:
> So, please, check if it's running:
> `# ps | grep mqttv4`

I got this
[image]

ther was no file to kill
[image]

But I could restart it and got a callback
[image]

captaincanada, 2023-11-18:
wow, it keeps the state now! Great 👍 
tomorrow I will test if it's still working after a reboot.

I need the cam at the moment, baby boy is sleeping now

ttarzan01, 2023-11-18:
I completely disconnected the camera from the power supply and restarted it. However, I keep getting the same output in the terminal when I try to start mqttv4.

captaincanada, 2023-11-18:
> wow, it keeps the state now! Great 👍 tomorrow I will test if it's still working after a reboot.
> 
> I need the cam at the moment, baby boy is sleeping now

could test it now and the problem is back.
it doesn't save the state

captaincanada, 2023-11-18:
why do I have to restart mqtt4 everytime now?

roleoroleo, 2023-11-18:
@ttarzan01
In your case, the cam is unable to connect to the mqtt broker.
[image]
Could you check the server parameters?

@captaincanada
In your case, the process doesn't start at boot (or it dies immediately).
Edit the file /tmp/sd/yi-hack/script/system.conf and redirect the log of the process.
From:
```
if [ "$HV" == "11" ] || [ "$HV" == "12" ]; then
    if [ "$MODEL_SUFFIX" != "y291ga" ] && [ "$MODEL_SUFFIX" != "y211ga" ]; then
        mqttv4 -t local &
    else
        mqttv4 &
    fi
else
    mqttv4 &
fi
```
To:

```
if [ "$HV" == "11" ] || [ "$HV" == "12" ]; then
    if [ "$MODEL_SUFFIX" != "y291ga" ] && [ "$MODEL_SUFFIX" != "y211ga" ]; then
        mqttv4 -t local > /tmp/mqttv4.log 2>&1 &
    else
        mqttv4 > /tmp/mqttv4.log 2>&1 &
    fi
else
    mqttv4 > /tmp/mqttv4.log 2>&1 &
fi
```

ttarzan01, 2023-11-18:
> Could you check the server parameters?

Where can I find them?

captaincanada, 2023-11-19:
> @captaincanada In your case, the process doesn't start at boot (or it dies immediately). Edit the file /tmp/sd/yi-hack/script/system.conf and redirect the log of the process. From:
>
> ```
> if [ "$HV" [...]

did it... 
I found System.sh but I'm sure that's the right
[image]

no change after reboot

the log file contains this
[image]

roleoroleo, 2023-11-19:
> Where can I find them?

Probably now I understand the problem: you don't use a mqtt broker.
Right?

roleoroleo, 2023-11-19:
@captaincanada 
There is a fatal error:
`Can't open mqueue /ipc_dispatch_1`
This shouldn't happen. Probably a problem in the start procedure of the hack.
Post your /tmp/sd/lower_half_init.sh.

ttarzan01, 2023-11-19:
Hmm, on index.html?page=mqtt MQTT ist enabled:
[image]

`Configuration file for mqttv4
# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.178.100
MQTT_PORT=1883

MQTT_CLIENT_ID=yicam-penny1

MQTT_USER=
MQTT_PASSWORD=

MQTT_PREFIX=yicam-penny1
MQTT_PREFIX_CONFIG=`

But on index.html?page=mqtt_adv is MQTT disabled:
[image]

captaincanada, 2023-11-19:
#!/bin/sh

### wifi 8188 ###
if [ "${enable_4g}" = "y" ];then
    echo "4g is running...."
else
    if [ -f /home/base/wifi/8188fu.ko ];then
        insmod /home/base/wifi/8188fu.ko
    elif [ -f /home/base/wifi/8189fs.ko ];then
        insmod /home/base/wifi/8189fs.ko
    elif [ -f /backup/ko/8188fu.ko ];then
        insmod /backup/ko/8188fu.ko
    elif [ -f /backup/ko/8189fs.ko ];then
        insmod /backup/ko/8189fs.ko
    elif [ -f /backup/ko/8192fu.ko ];then
        insmod /backup/ko/8192fu.ko
    elif [ -f /backup/ko/ssv6x5x.ko ];then
        if [ "${SUFFIX}" = "d071qp"  ];then
            if [ -f /home/base/firmware/ssv6x5x/ssv6152-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6152-wifi.cfg"
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi	
        else
            if [ -f /home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg"
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi
        fi
    fi
fi

echo "--------------------------insmod sensor--------------------------"
insmod /home/base/ko/videobuf2-core.ko
insmod /home/base/ko/videobuf2-memops.ko
insmod /home/base/ko/videobuf2-dma-contig.ko
insmod /home/base/ko/videobuf2-v4l2.ko
insmod /home/base/ko/vin_io.ko

if [ "${SUFFIX}" = "b091qp" ];then
    insmod /backup/ko/cam_sensor.ko
    insmod /home/base/ko/vin_v4l2.ko ccm0=$SENSOR_DRIVE_NAME i2c0_addr=$SENSOR_ADDR
else
    insmod /home/base/ko/cam_sensor.ko
    insmod /home/base/ko/vin_v4l2.ko
fi

if [ -f /home/base/ko/icplus.ko ];then
    insmod /home/base/ko/icplus.ko
elif [ -f /backup/ko/icplus.ko ];then
    insmod /backup/ko/icplus.ko
fi

if [ -f /home/base/ko/sunxi_gpadc.ko ];then
    insmod /home/base/ko/sunxi_gpadc.ko
fi

#/home/app/script/factory_test.sh

#echo "MTK 7601" > /tmp/MTK
#echo /tmp/sd/core.exe[%e].pid[%p].sig[%s] > /proc/sys/kernel/core_pattern

sleep 1
ifconfig lo up

ifconfig ${NETWORK_IFACE} up
ethmac=d2:`ifconfig ${NETWORK_IFACE} |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#if [ "${enable_4g}" = "y" ];then
#    ifconfig usb0 up
#    ethmac=d2:`ifconfig usb0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#else
#    ifconfig wlan0 up
#    ethmac=d2:`ifconfig wlan0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#fi

ifconfig eth0 hw ether $ethmac
a=1
if [ "${SUFFIX}" = "b111qp" ] || [ "${SUFFIX}" = "b101qp" ] || [ "${SUFFIX}" = "b092qp" ] || [ "${SUFFIX}" = "b091qp" ] || [ "${SUFFIX}" = "q321br_aldz_3m" ]; then
    while ( ! ifconfig eth0 up)
    do
        echo "ifconfig eth0 up failed"
        let a++
        if [ $a -eq 10 ]; then
            break
        fi
    done
else
    ifconfig eth0 up
fi

HOMEVER=$(cat /home/homever)
HV=${HOMEVER:0:2}

if [ "$HV" == "11" ] || [ "$HV" == "12" ]; then
    ln -s /home/model/BodyVehicleAnimal3.model /tmp/BodyVehicleAnimal3.model
fi

echo "============================================= home low_half_init.sh... ========================================="
echo "============================================= begin to start app... ========================================="
cd /home/app
if [ -f /home/app/property ];then
    ./property &
fi
#./log_server &

if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
	/tmp/sd/Factory/config.sh
	exit
fi

if [ "$HV" == "11" ] || [ "$HV" == "12" ]; then
    if [ -f "/tmp/sd/log_tools.tar.gz" ];then
        echo "run log_tools start."
        if [ ! -d /tmp/sd/log_tools ];then
            cd /tmp/sd
            mkdir log_tools
        fi
        cd /tmp/sd
        tar -zxvf log_tools.tar.gz -C /tmp/sd/log_tools
        chmod +x /tmp/sd/log_tools/run_log_app.sh
        source /tmp/sd/log_tools/run_log_app.sh
        cd -
        echo "run log_tools end."
        #exit
    fi
fi

mount --bind /tmp/sd/yi-hack/script/wifidhcp.sh /home/app/script/wifidhcp.sh
mount --bind /tmp/sd/yi-hack/script/wifidhcp.sh /backup/tools/wifidhcp.sh

LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./rtmp &
#./watch_process &

chmod 777 /tmp/sd/debug.sh
if [ -f "/tmp/sd/debug.sh" ]; then
    echo "calling /tmp/sd/debug.sh"
    sh /tmp/sd/debug.sh &
fi

chmod 755 /tmp/sd/yi-hack/script/system.sh
sh /tmp/sd/yi-hack/script/system.sh &

roleoroleo, 2023-11-19:
@ttarzan01
mqttv4 exits with a connection error related to your mqtt broker.
So, check that the broker is running on this IP 192.168.178.100 and the connection is not filtered.

roleoroleo, 2023-11-19:
@captaincanada
lower_half_init.sh is ok.
The error `Can't open mqueue /ipc_dispatch_1` can be a problem with dispatch.
Please, try to reinstall the hack from scratch: format again the sd card and decompress the last released version.

captaincanada, 2023-11-20:
good morning. 

I reinstalled the whole cam and it works now👍

what I noticed is that I need the port to access the web service anymore...

@ttarzan01 
I suggest to do the same.
I configured network credentials and there was no problem refreshing the hack.
It took about 5min

@roleoroleo 
to use it with the home assistant plug in, do I have to change some settings of the cam?

maybe I have overwritten something the last time i tried to set it up for home assistant...

thx a lot for the great support!!!

roleoroleo, 2023-11-20:
The port depends on your configuration.
The old version of the hack used port 8080.
Some version ago I changed and now the standard port is 80.
But finally, it depends on your configuration.

To use with ha, you need to enable mqtt and you need a mqtt broker (i.e. mqtt components fot ha).
When you start the wizard to configure the cam, insert the http address of the cam and the port where it resides.
No other requirements.

ttarzan01, 2023-11-21:
Ok, I have completely reinstalled the hack. Despite everything, I don't see any change. The status on 'cgi-bin/get_configs.sh?conf=camera' does not change when I switch off the camera. I use the web service 'cgi-bin/camera_settings.sh?switch_on=no' from HA. The camera goes off, but the status never changes. 'service.sh?name=mqtt&action=status' tells me that MQTT is running.

ps | grep mqttv4 says:

```
1402 root       792 S    mqttv4
8466 root       980 S    grep mqttv4
```

captaincanada, 2023-11-21:
did you reboot the cam? I forgot to do that the last time and was wondering...

ttarzan01, 2023-11-21:
> did you reboot the cam?

Yes. I have rebooted the camera. After installing the hack, I disconnected the camera from the power supply and reconnected it.

captaincanada, 2023-11-21:
one more question about the HA integration:

I saw that the default port on the mqtt settings (?page=mqtt) is set to 1883.
[image]

this wasn't configured on my broker in Home Assistant. 
After activate this "hidden" port on the Broker and setting a user/pw on the web interface (?page=mqtt) the integration was showing "connected" as state.

[image]

[image]

But why do I have an disconnected user <unknown> in my log?
it tries to connect regularly...
[image]

roleoroleo, 2023-11-21:
Are you using different devices with the same client id?
[image]

captaincanada, 2023-11-21:
when i configured the integration the client id was set automatically.

yi_hack_a2_4898ff

i didn't use this id for my mqtt configuration on the web client... Do i have to use this name that was created?

There is no other Yi cam in the Network.

roleoroleo, 2023-11-21:
In the image above, client id is set to "yi-cam".
If you have 2 devices with the same id, the second one will not login to the broker.

captaincanada, 2023-11-21:
OK, so that means in the mqtt config must be the same like in Home Assistant (yi_hack_a2_4898ff in my case)?!

i edited the name in Home Assistant, to Yi_babyphone ant set this name in the web config... 
then this appeared in the log
[image]

i changed the name now of the entity (its called Yi_babyphone now and i couldn't change the

roleoroleo, 2023-11-21:
Please, check again the log of mqttv4.

captaincanada, 2023-11-21:
[image]

roleoroleo, 2023-11-22:
I mean the log: the procedure explained here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/664#issuecomment-1817607252

captaincanada, 2023-11-22:
[image]

[image]

the  unknown User is a 172.30... IP
what could that be?
it is still in the log shown above

roleoroleo, 2023-11-23:
The log seems ok now.
Now you should try to remove the cam from ha and add it again.

I don'y know what could be the IP 172.30..
Maybe ha?

ttarzan01, 2023-11-24:
It seems that my problem is slightly different from @captaincanada . I don't need all the functionalities that MQTT provides in Home Assistant. I just switch the camera on or off via the web service. That still works.

I have also queried the status of the camera via the web service, whereby I have simply evaluated the JSON, which I have sent via "... cgi-bin/get_configs.sh?conf=camera". And here is the problem: the status never changes in the JSON. It is always "on".

[15 more replies]

closed 2024-04-17 by github-actions[bot]