roleoroleo/yi-hack-Allwinner-v2 #665: ip-cam connection problem

gregoo31, 2023-07-10:
Good morning,

I bought a yi outdoor 1080p camera a few weeks ago:

- H30GA v1.4
- production model YHS.3017
- DID: RFUSY59X60XJAU220620
- SN: RH30GAEUW320000400
- date: 2023.02

The installation phase of the files seem to be installed correctly in the camera (I have "Factory.done", the card is formatted by the camera in fat32...), it reconnects to the wifi but I cannot join it with its "http://ip-cam" and "http://ip-cam:8080" and so I don't have access to the "website" of the camera.

I tried several microSD cards, I tried other hacks but it's always the same...
Is this a problem with the new models ? or am I too dumb ;-)

If you have an idea, I'm interested ! THANKS

roleoroleo, 2023-07-10:
Telnet is open but ssh no.
So the hack is ok but the new processes don't start.
Please enter with telnet (root no password) and run these commands:
```
ps
ls -laR /tmp/sd/yi-hack
cat /tmp/sd/yi-hack/script/system.sh
```

gregoo31, 2023-07-10:
Thank you Roleoroleo,
I did the commands you gave me, but it seems that the file is missing?
(personally I don't understand, this goes way beyond my computer programming skills)

(none) login: root

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@(none):~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  207 root         0 SW   [oom_reaper]
  208 root         0 SW<  [writeback]
  210 root         0 SW   [kcompactd0]
  211 root         0 SW<  [crypto]
  212 root         0 SW<  [bioset]
  214 root         0 SW<  [kblockd]
  219 root         0 SW   [kworker/u2:2]
  256 root         0 SW   [sys_user]
  263 root         0 SW   [kworker/0:1]
  264 root         0 SW<  [cfg80211]
  271 root         0 SW<  [watchdogd]
  285 root         0 SW<  [spi0]
  295 root         0 SW   [kswapd0]
  379 root         0 SW<  [bioset]
  384 root         0 SW<  [bioset]
  389 root         0 SW<  [bioset]
  394 root         0 SW<  [bioset]
  399 root         0 SW<  [bioset]
  404 root         0 SW<  [bioset]
  409 root         0 SW<  [bioset]
  414 root         0 SW<  [bioset]
  419 root         0 SW<  [bioset]
  451 root         0 SW   [irq/302-sunxi-m]
  453 root         0 SW   [irq/166-sdc0 cd]
  454 root         0 SW   [irq/303-sunxi-m]
  457 root         0 SW   [kworker/0:2]
  525 root         0 SW<  [kworker/0:1H]
  526 root         0 SW   [kworker/0:3]
  528 root         0 SW<  [bioset]
  529 root         0 SW   [mmcqd/0]
  545 root         0 SWN  [jffs2_gcd_mtd4]
  550 root       980 S    -/bin/sh
  582 root       988 S    /usr/sbin/telnetd
  590 root         0 SW   [ksdioirqd/mmc1]
  597 root         0 SW   [RTW_XMIT_THREAD]
  598 root         0 SW   [RTW_CMD_THREAD]
  599 root         0 SW   [RTWHALXT]
  641 root      1172 S    ./dispatch
  686 root      1400 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
  714 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
 1557 root       980 S    -ash
 1629 root       980 R    ps
root@(none):~# ls -laR /tmp/sd/yi-hack
/tmp/sd/yi-hack:
drwxr-xr-x    2 root     root          8192 Jun 21 16:05 .
drwxr-xr-x    7 root     root          8192 Jul 10 13:21 ..
-rwxr-xr-x    1 root     root             0 Jun 21 16:05 fw_upgrade_in_progress
root@(none):~# cat /tmp/sd/yi-hack/script/system.sh
cat: can't open '/tmp/sd/yi-hack/script/system.sh': No such file or directory
root@(none):~#

roleoroleo, 2023-07-10:
The content of the sd card is not ok.
Please download again the last released archive and decompress again on your sd card.

closed 2023-07-10 by gregoo31