roleoroleo/yi-hack-Allwinner-v2 #668: Can't upload to FTP

alfiriel, 2023-07-11:
My Settings:
**Events FTP Upload**
- FTP Upload: Enabled
- FTP Delete: Disabled
- Create directory tree: Disabled

**FTP Options**
- FTP Host: 192.168.xx.xx
- FTP Directory: cameras
- FTP Username: cam
- FTP Password: ***

When I connect via FileZilla, I can upload files in the cameras folder of the ftp server without problems
When I connect via SSH and run  /tmp/sd/yi-hack/script/ftppush.sh cron, the output is:

>root@cam01:~#  /tmp/sd/yi-hack/script/ftppush.sh cron
>2023-07-11 [06-31-08] [INFO] === SERVICE START ===
>2023-07-11 [06-31-08] [INFO] checkFiles
>ftpput: unexpected server response to STOR: 553 Users may not upload files
>2023-07-11 [06-31-09] [ERROR] uploadToFtp: ftpput FAILED.
>2023-07-11 [06-31-09] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2023Y07M06D13H/E201M57S03.mp4]. Retrying in 45 s.
>2023-07-11 [06-31-09] [INFO] === SERVICE STOPPED ===

Double checked username an password, No special Chars in User Password.

Can anyone help?

roleoroleo, 2023-07-11:
Try the command ftpput manually.

alfiriel, 2023-07-11:
Used this ftpput command, hope it is correct (ChatGPT helped me ;))
>ftpput -v -u cam -p PASSWORD 192.168.xx.xx cameras/E201M57S03.mp4 /tmp/sd/record/2023Y07M06D13H/E201M57S03.mp4

Same error message:
>ftpput: unexpected server response to STOR: 553 Users may not upload files

roleoroleo, 2023-07-11:
I tested your config on my cam and it works properly.
Please, check the permissions of your file server.

alfiriel, 2023-07-11:
It is working fine with filezilla:
[image]
I can create folders and upload files.

My FTP Server is a WD MyCloud EX2, there I can give read or write permission via webUI. The user cam has read and write permissions.

alfiriel, 2023-07-11:
Hi, again.
Seems that I have found the issue.
In the yi-hack FTP Options, I added a / in front of the FTP directory:
[image]

When I try the `/tmp/sd/yi-hack/script/ftppush.sh cron` command via SSH, I receive:

>root@cam01:~# /tmp/sd/yi-hack/script/ftppush.sh cron
>2023-07-11 [10-57-32] [INFO] === SERVICE START ===
>2023-07-11 [10-57-32] [INFO] checkFiles
>2023-07-11 [10-57-37] [INFO] checkFiles: uploadToFtp SUCCEEDED - [/tmp/sd/record/2023Y07M11D07H/E207M02S58.mp4].
>[ERROR] uploadToFtp: File not found.
>2023-07-11 [10-57-38] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2023Y07M11D07H/E208M00S60.mp4]. >Retrying in 45 s.
>2023-07-11 [10-57-39] [INFO] === SERVICE STOPPED ===

And I have a lot of files now on my ftp. GREAT! 💯 
I do not know why it stopped on this one file.

But ftp Problem solved.

Thank you for your fast reaction and support @roleoroleo

closed 2023-07-11 by alfiriel