roleoroleo/yi-hack-Allwinner-v2 #670: Reboot loop when cloud disabled?

dbrb2, 2023-07-15:
I have y211ga_0.27 installed onto the following device: 

Model Suffix | y211ga
Serial Number | QFUSY62FNZEK8E230310
Base version 9.0.35.00_202111291330

All seems to work - except that if I disable cloud entirely, the unit resets every ~60 seconds 

Is this some kind of watchdog kicking in on the camera? I can acheive the same results by blocking the camera from the world on my firewall - and that seems to avoid the reboot issue entirely....

roleoroleo, 2023-07-16:
Please, try to run ps and top commands while the cam is running.

dbrb2, 2023-07-16:
Here you are - this is in the stable config (cloud enabled) 
Getting the same with cloudd disabled is tricky because of the restarts, but I'll see what I can get 

[ps_cloud_disabled.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12064073/ps_cloud_disabled.log)
[top_cloud_enabled.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12064074/top_cloud_enabled.log)

roleoroleo, 2023-07-16:
It's not the first time I've encountered this problem.
If you compare the logs, the difference is the rmm process.
I don't know why but it dies and the wathcdog restarts the cam.

If you want to try, you could increase the sleep time from 6 to 10 in system.sh (line 203).

dbrb2, 2023-07-16:
Unfortunately changing that sleep to 10 didn't change the behaviour:

64 bytes from 192.168.1.17: icmp_seq=11 ttl=64 time=17.5 ms
64 bytes from 192.168.1.17: icmp_seq=12 ttl=64 time=8.01 ms
64 bytes from 192.168.1.17: icmp_seq=13 ttl=64 time=10.8 ms
64 bytes from 192.168.1.17: icmp_seq=14 ttl=64 time=11.5 ms
64 bytes from 192.168.1.17: icmp_seq=15 ttl=64 time=8.47 ms
64 bytes from 192.168.1.17: icmp_seq=16 ttl=64 time=8.40 ms
64 bytes from 192.168.1.17: icmp_seq=17 ttl=64 time=6.33 ms
64 bytes from 192.168.1.17: icmp_seq=18 ttl=64 time=15.1 ms
64 bytes from 192.168.1.17: icmp_seq=19 ttl=64 time=9.71 ms
64 bytes from 192.168.1.17: icmp_seq=20 ttl=64 time=11.4 ms
64 bytes from 192.168.1.17: icmp_seq=21 ttl=64 time=7.00 ms
64 bytes from 192.168.1.17: icmp_seq=22 ttl=64 time=9.45 ms
64 bytes from 192.168.1.17: icmp_seq=23 ttl=64 time=4.79 ms
64 bytes from 192.168.1.17: icmp_seq=24 ttl=64 time=9.32 ms
64 bytes from 192.168.1.17: icmp_seq=25 ttl=64 time=11.0 ms
64 bytes from 192.168.1.17: icmp_seq=26 ttl=64 time=12.6 ms
64 bytes from 192.168.1.17: icmp_seq=27 ttl=64 time=12.6 ms
64 bytes from 192.168.1.17: icmp_seq=28 ttl=64 time=8.39 ms
64 bytes from 192.168.1.17: icmp_seq=29 ttl=64 time=11.1 ms
64 bytes from 192.168.1.17: icmp_seq=30 ttl=64 time=9.18 ms
64 bytes from 192.168.1.17: icmp_seq=31 ttl=64 time=18.1 ms
64 bytes from 192.168.1.17: icmp_seq=32 ttl=64 time=10.7 ms
64 bytes from 192.168.1.17: icmp_seq=33 ttl=64 time=6.77 ms
64 bytes from 192.168.1.17: icmp_seq=34 ttl=64 time=8.52 ms
64 bytes from 192.168.1.17: icmp_seq=35 ttl=64 time=12.0 ms
64 bytes from 192.168.1.17: icmp_seq=36 ttl=64 time=8.35 ms
64 bytes from 192.168.1.17: icmp_seq=37 ttl=64 time=7.65 ms
64 bytes from 192.168.1.17: icmp_seq=38 ttl=64 time=15.0 ms
64 bytes from 192.168.1.17: icmp_seq=39 ttl=64 time=25.0 ms
64 bytes from 192.168.1.17: icmp_seq=40 ttl=64 time=12.3 ms

RESTARTS

64 bytes from 192.168.1.17: icmp_seq=61 ttl=64 time=115 ms
64 bytes from 192.168.1.17: icmp_seq=62 ttl=64 time=8.32 ms
64 bytes from 192.168.1.17: icmp_seq=63 ttl=64 time=8.18 ms
64 bytes from 192.168.1.17: icmp_seq=64 ttl=64 time=7.29 ms
64 bytes from 192.168.1.17: icmp_seq=65 ttl=64 time=6.05 ms
64 bytes from 192.168.1.17: icmp_seq=66 ttl=64 time=6.76 ms
64 bytes from 192.168.1.17: icmp_seq=67 ttl=64 time=9.15 ms
64 bytes from 192.168.1.17: icmp_seq=68 ttl=64 time=8.38 ms
64 bytes from 192.168.1.17: icmp_seq=69 ttl=64 time=6.30 ms
64 bytes from 192.168.1.17: icmp_seq=70 ttl=64 time=16.5 ms
64 bytes from 192.168.1.17: icmp_seq=71 ttl=64 time=6.61 ms
64 bytes from 192.168.1.17: icmp_seq=72 ttl=64 time=8.24 ms
64 bytes from 192.168.1.17: icmp_seq=73 ttl=64 time=7.35 ms
64 bytes from 192.168.1.17: icmp_seq=74 ttl=64 time=9.32 ms
64 bytes from 192.168.1.17: icmp_seq=75 ttl=64 time=8.77 ms
64 bytes from 192.168.1.17: icmp_seq=76 ttl=64 time=8.76 ms
64 bytes from 192.168.1.17: icmp_seq=77 ttl=64 time=10.3 ms
64 bytes from 192.168.1.17: icmp_seq=78 ttl=64 time=10.9 ms
64 bytes from 192.168.1.17: icmp_seq=79 ttl=64 time=12.5 ms
64 bytes from 192.168.1.17: icmp_seq=80 ttl=64 time=7.50 ms
64 bytes from 192.168.1.17: icmp_seq=81 ttl=64 time=6.19 ms
64 bytes from 192.168.1.17: icmp_seq=82 ttl=64 time=8.78 ms
64 bytes from 192.168.1.17: icmp_seq=83 ttl=64 time=6.25 ms
64 bytes from 192.168.1.17: icmp_seq=84 ttl=64 time=17.9 ms
64 bytes from 192.168.1.17: icmp_seq=85 ttl=64 time=8.39 ms
64 bytes from 192.168.1.17: icmp_seq=86 ttl=64 time=9.37 ms
64 bytes from 192.168.1.17: icmp_seq=87 ttl=64 time=8.02 ms
64 bytes from 192.168.1.17: icmp_seq=88 ttl=64 time=12.4 ms
64 bytes from 192.168.1.17: icmp_seq=89 ttl=64 time=6.45 ms

RESTARTS

closed 2023-08-18 by roleoroleo

vekexasia, 2023-10-12:
I'm having the same issue but i've seen it's been marked as completed by @roleoroleo ... anything else i can provide to help debug?

PR #685 (2023-08-18): Add "sleep 1" before rmm run
Without this wait, the rmm process fails to start on y291ga with the "Disabled cloud" option enabled. Fixes #670.

-----------------

This feels more like a workaround rather than a fix though.
For some reasons, rmm fails to start if started without the wait.

Here you can find the output of rmm when started manually after the boot (after having disabled the WD's reboot) and the failed one when starting it within `system.sh` (without the wait).

[rmm_fail.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12233338/rmm_fail.txt)
[rmm_success.txt](https:// [... cut]
commit c3f4dd48 (2023-08-18): Add "sleep 1" before rmm run

Without this wait, the rmm process fails to start on y291ga with the
"Disabled cloud" option enabled. Fixes #670.