roleoroleo/yi-hack-Allwinner-v2 #675: Unable to download firmware file

rt400, 2023-07-18:
Hi,
I have 2 camera r35gb, r30gb.
Both have error "Unable to download firmware file" when try to update firmware from web.
thanks for your help...

roleoroleo, 2023-07-18:
Here the piece of code:
```
        /tmp/sd/yi-hack/usr/bin/wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/$LATEST_FW/${MODEL_SUFFIX}_${LATEST_FW}.tgz
        if [ ! -f ${MODEL_SUFFIX}_${LATEST_FW}.tgz ]; then
            printf "Content-type: text/html\r\n\r\n"
            printf "Unable to download firmware file."
            exit
        fi
```
wget fails, try to run it manually inside the cam:
/tmp/sd/yi-hack/usr/bin/wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.9/r30gb_0.2.9.tgz

rt400, 2023-07-18:
can't run the commands, keep said "Read-only file system"

roleoroleo, 2023-07-19:
Maybe your sd card is mounted in read-only mode.
Probably a little corruption.
Extract from the cam and try to repair it with a PC.

closed 2023-11-17 by stale[bot]