roleoroleo/yi-hack-Allwinner-v2 #676: Yi Cam Outdoor 1080p

chross, 2023-07-19:
Hi!
My cam is working with this hack for some time but then crashes and wont reboot until I power cycle it.
I have already disables all services but RTSP low res and Screenshot, also I enables swap and bought a new brand SD card (4GB) which is FAT32 formatted.
I constantly fetch the stream with my NAS solution and sometimes fetch an additional low res snapshot, no recordings are made on camera.

roleoroleo, 2023-07-19:
Is the crash periodic?
Or random?

chrbar11, 2023-07-24:
Hi @chross
I meet some troubles with my Yi Outdoor 1080p.
I'd found a [discussion](https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/561) about that which I just answered to try to get advice.
Hope this can help you too :)

roleoroleo, 2023-07-24:
> What is the use of the alternative RTSP server? Maybe switching RTSP helps?

It's a much simpler server from a code point of view.
Lighter but also with less features.
Some users using it with NVR if they have problem with main RTSP service.

closed 2023-11-21 by stale[bot]