roleoroleo/yi-hack-Allwinner-v2 #677: add support for wireguard

theguly, 2023-07-20:
Hello, i have quickly read the source of the hack and i believe it is technically possible to add wireguard to the hack addons, if wireguard can be compiled on the target architecture against musl (which i did not investigated yet).

Are you eventually interested in a PR ? I would like to be able to reach my camera directly through my wg vpn and i can probably spend some of my spare time on this. Of course the quality of the "deliverable" will be different if it will be deployed only on my camera/you will accept a PR :)

Thanks

roleoroleo, 2023-07-21:
Yes. I'm interested in a PR.
I could test it and add the "web gui" part.

theguly, 2023-07-25:
i believe i've been overly optimistic about the kernel module, this will take more time than expected :D

will update as soon as i have news

theguly, 2023-10-23:
you're totally right. i bricked the camera (no, removing the SD doesn't fix :/ ) and a huge workload prevented me to even try to unbrick.

anyway, i have had issues while trying to load the compiled modules on camera so probably i have to restart from the very beginning with the toolchain.

it will not happen very soon unfortunately

theguly, 2023-10-27:
> I think the problem is that you miss the kernel config. The vpn kernel module is very dependent on how you configured the kernel and I think that, without having the configuration, it is very difficult to make it work.

fair enough, now that i think about it i did not found a way to have the kernel config from my camera: nothing in /boot of course, and nothing in /proc as well. did you found a way to dump kernel config by any chance? do you think this is a showstopper?

to be honest, i think i did not try checking /lib/modules/$(uname -r)/build/.config or extract-ikconfig

closed 2024-02-25 by stale[bot]