roleoroleo/yi-hack-Allwinner-v2 #678: Black screen sometimes

cbroissa-github, 2023-07-23:
Hello,

I have several cameras (1080 Home IFUS, RFUS, Outdoor IFUS), with the latest release (same problem with old releases).
Sometimes I have a black screen, and after a refresh, the image is displayed, and few seconds after another black screen.
It’s the same for all cameras, with wifi strength from 60% to 80%.
I use webrtc and rtsp url.

Thank you for your job.

liebefuroscar, 2023-09-25:
Same in my cameras

roleoroleo, 2023-09-26:
Could you test the cam with another client?
For example with vlc or ffmpeg?

liebefuroscar, 2023-09-26:
When I play the stream on VLC it plays normally.
On Frigate has some lag, but is working aprently fine.
Every random minute stream stop working and black image happens. When im seeing black image on Frigate, i´ll try to play the stream via VLC and nothing happen.

Ill tried with both streams enabled, only HQ, only LQ, disabling all features like FTP, Snapshot...
The only "stable" I found is use "RTSP alternative server" but plays without audio.

roleoroleo, 2023-09-26:
What's your cam suffix?

liebefuroscar, 2023-09-26:
Y291ga, Yi 1080p Home RFUS using Base Version	12.0.35.8_202211241029 and 0.2.9 firmware

roleoroleo, 2023-09-30:
I made a lot of improvements in the RTSP server.
Do you want to try a beta?

liebefuroscar, 2023-09-30:
Yes. Very grateful to be able to help the development

cbroissa-github, 2023-09-30:
I can test it too.
Thank you.

roleoroleo, 2023-09-30:
Here it is:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12774966/rRTSPServer.gz)

cbroissa-github, 2023-10-01:
How to install the file ?

liebefuroscar, 2023-10-02:
But the error keeps happening to me.
When both RTSP streams (HQ and LQ) and aac audio are enabled, I only get video on the HQ stream and a black screen on the LQ. And I don't have audio on either of them.

When I enable "RTSP alternative server " I get video on both, but audio on neither

roleoroleo, 2023-10-09:
> How to install the file ?

Rename the old binary in /tmp/sd/yi-hack/bin and copy the new one.

> But the error keeps happening to me.
> When both RTSP streams (HQ and LQ) and aac audio are enabled, I only get video on the HQ stream and a black screen on the LQ. And I don't have audio on either of them.

Please, try to run ffmpeg and post the output:
`ffmpeg -i rtsp://IP-CAM/ch0_0.h264 test.mp4`
Try tcp too:
`ffmpeg -rtsp_transport tcp -i rtsp://IP-CAM/ch0_0.h264 test.mp4`

liebefuroscar, 2023-10-09:
tell me how or where?
By the way, much more stable using both streams with new "rRTSPServer.gz" (but wt audio with alternativer RTSP server for me)

roleoroleo, 2023-10-12:
> tell me how or where?

Use a PC with linux or Windows and install ffmpeg.
Then run the commands above.

liebefuroscar, 2023-10-15:
ffmpeg -i rtsp://IP-CAM/ch0_0.h264 test.mp4 :

ffmpeg version 2023-10-12-git-a7663c9604-full_build-www.gyan.dev Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.0 (Rev2, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --pkg-config=pkgconf --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-bzlib --enable-lzma --enable-libsnappy --enable-zlib --enable-librist --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-libbluray --enable-libcaca --enable-sdl2 --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libdavs2 --enable-libuavs3d --enable-libzvbi --enable-librav1e --enable-libsvtav1 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs2 --enable-libxvid --enable-libaom --enable-libjxl --enable-libopenjpeg --enable-libvpx --enable-mediafoundation --enable-libass --enable-frei0r --enable-libfreetype --enable-libfribidi --enable-libharfbuzz --enable-liblensfun --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libvpl --enable-libshaderc --enable-vulkan --enable-libplacebo --enable-opencl --enable-libcdio --enable-libgme --enable-libmodplug --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libshine --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libcodec2 --enable-libilbc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-ladspa --enable-libbs2b --enable-libflite --enable-libmysofa --enable-librubberband --enable-libsoxr --enable-chromaprint
  libavutil      58. 27.100 / 58. 27.100
  libavcodec     60. 30.102 / 60. 30.102
  libavformat    60. 15.100 / 60. 15.100
  libavdevice    60.  2.101 / 60.  2.101
  libavfilter     9. 11.100 /  9. 11.100
  libswscale      7.  4.100 /  7.  4.100
  libswresample   4. 11.100 /  4. 11.100
  libpostproc    57.  2.100 / 57.  2.100
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000021e66696cc0] decode_slice_header error
[h264 @ 0000021e66696cc0] no frame!
[h264 @ 0000021e66696cc0] non-existing PPS 0 referenced
[... 146 lines cut ...]
[h264 @ 0000021e66696cc0] no frame!
Input #0, rtsp, from 'rtsp://user:pass@MYIP-CAM/ch0_0.h264':
  Duration: N/A, start: 2.868000, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 39 tbr, 90k tbn
File 'test.mp4' already exists. Overwrite? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[libx264 @ 0000021e670729c0] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 0000021e670729c0] profile High, level 4.0, 4:2:0, 8-bit
[libx264 @ 0000021e670729c0] 264 - core 164 r3144 5a9dfdd - H.264/MPEG-4 AVC codec - Copyleft 2003-2023 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.0 [... cut ...]
[rtsp @ 0000021e66679e00] max delay reached. need to consume packet=5243.8kbits/s dup=0 drop=5 speed=0.701x
[rtsp @ 0000021e66679e00] RTP: missed 83 packets
[h264 @ 0000021e66ca0780] error while decoding MB 93 9, bytestream -13
[h264 @ 0000021e66ca0780] concealing 7036 DC, 7036 AC, 7036 MV errors in I frame
[vist#0:0/h264 @ 0000021e6672f580] corrupt decoded frame.70 bitrate=1553.6kbits/s dup=0 drop=5 speed=2.25x
frame= 1411 fps= 21 q=28.0 size=   20736kB time=00:01:10.45 bitrate=2411.2kbits/s dup=21 drop=8 speed=1.04x

liebefuroscar, 2023-10-15:
ffmpeg -rtsp_transport tcp -i rtsp://IP-CAM/ch0_0.h264 test.mp4:
ffmpeg version 2023-10-12-git-a7663c9604-full_build-www.gyan.dev Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.0 (Rev2, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --pkg-config=pkgconf --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-bzlib --enable-lzma --enable-libsnappy --enable-zlib --enable-librist --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-libbluray --enable-libcaca --enable-sdl2 --enable-libaribb24 --enable-libaribcaption --enable-libdav1d --enable-libdavs2 --enable-libuavs3d --enable-libzvbi --enable-librav1e --enable-libsvtav1 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs2 --enable-libxvid --enable-libaom --enable-libjxl --enable-libopenjpeg --enable-libvpx --enable-mediafoundation --enable-libass --enable-frei0r --enable-libfreetype --enable-libfribidi --enable-libharfbuzz --enable-liblensfun --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libvpl --enable-libshaderc --enable-vulkan --enable-libplacebo --enable-opencl --enable-libcdio --enable-libgme --enable-libmodplug --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libshine --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libcodec2 --enable-libilbc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-ladspa --enable-libbs2b --enable-libflite --enable-libmysofa --enable-librubberband --enable-libsoxr --enable-chromaprint
  libavutil      58. 27.100 / 58. 27.100
  libavcodec     60. 30.102 / 60. 30.102
  libavformat    60. 15.100 / 60. 15.100
  libavdevice    60.  2.101 / 60.  2.101
  libavfilter     9. 11.100 /  9. 11.100
  libswscale      7.  4.100 /  7.  4.100
  libswresample   4. 11.100 /  4. 11.100
  libpostproc    57.  2.100 / 57.  2.100
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[... 20 lines cut ...]
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
[h264 @ 0000017db20f6bc0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0000017db20f6bc0] decode_slice_header error
[h264 @ 0000017db20f6bc0] no frame!
Input #0, rtsp, from 'rtsp://user:pass@MYIP-CAM/ch0_0.h264':
  Duration: N/A, start: 1.418000, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 40 tbr, 90k tbn
File 'test.mp4' already exists. Overwrite? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[libx264 @ 0000017db2a7e040] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[... cut ...] 60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'test.mp4':
  Metadata:
    encoder         : Lavf60.15.100
  Stream #0:0: Video: h264 (avc1 / 0x31637661), yuvj420p(pc, bt709, progressive), 1920x1080, q=2-31, 20 fps, 10240 tbn
    Metadata:
      encoder         : Lavc60.30.102 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: N/A
frame=  746 fps= 21 q=28.0 size=   10752kB time=00:00:37.20 bitrate=2367.8kbits/s dup=8 drop=9 speed=1.06x

roleoroleo, 2023-10-16:
Very strange.
Maybe your cam has a different format.
I need a copy of the file /dev/shm/fshare_frame_buf to check the stream.

liebefuroscar, 2023-10-16:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12917217/fshare_frame_buf.zip)

liebefuroscar, 2023-10-17:
72 % is the worst of my 4 cameras using Yicam firmware

roleoroleo, 2023-10-19:
Please, another try.
Run rRTSPServer manually with the log.
`rRTSPServer -m y291ga -r both -a no -p 554 -d 15`
The log level could be 7 or 15, it depends on the version you are running.
Check the help:
`rRTSPServer --help`

When the process is started, run ffmpeg on the other side.

liebefuroscar, 2023-10-19:
I tried with both, d 15 and d 7 and this is what happen

root@Soco:~# killall rRTSPServer
killall: rRTSPServer: no process killed
root@Soco:~# rRTSPServer -m y291ga -r both -a no -p 554 -d 15
1697714144726: the size of the buffer is 1786224
1697714144733: capture - mapping file fshare_frame_buf, size 1786224, to b6bc300                                                                                                                                                             0
1697714144745: capture - closing the file fshare_frame_buf
1697714144745: capture - starting capture main loop
1697714144746: capture - buf_idx_end == buf_idx_end_prev
1697714144757: capture - buf_idx_end == buf_idx_end_prev
1697714144767: capture - ! n > 1
1697714144777: capture - warning - 7166 frame(s) lost
1697714144777: aac in - warning - 1960 AAC frame(s) lost - frame_counter: 1960 -                                                                                                                                                              frame_counter_last_valid: -1
1697714144777: h264/aac in - frame_len: 288 - cb_current->size: 0
1697714144777: h264/aac in - error - frame size exceeds buffer size
1697714144802: h26x in - warning - 2601 low res frame(s) lost - frame_counter: 2                                                                                                                                                             601 - frame_counter_last_valid: -1
1697714144803: h26x in - warning - 2604 high res frame(s) lost - frame_counter:                                                                                                                                                              2604 - frame_counter_last_valid: -1
1697714144828: h264/aac in - frame_len: 291 - cb_current->size: 0
1697714144828: h264/aac in - error - frame size exceeds buffer size
1697714144853: h26x in - frame detected - frame_len: 848 - frame_counter: 2602 -                                                                                                                                                              frame_counter_last_valid: 2601 - resolution: 360
1697714144878: h26x in - frame detected - frame_len: 2930 - frame_counter: 2605                                                                                                                                                              - frame_counter_last_valid: 2604 - resolution: 1080
1697714144878: h264/aac in - frame_len: 298 - cb_current->size: 0
1697714144878: h264/aac in - error - frame size exceeds buffer size
1697714144904: h26x in - frame detected - frame_len: 855 - frame_counter: 2603 -                                                                                                                                                              frame_counter_last_valid: 2602 - resolution: 360
1697714144929: h26x in - frame detected - frame_len: 2869 - frame_counter: 2606                                                                                                                                                              - frame_counter_last_valid: 2605 - resolution: 1080
1697714144954: h26x in - frame detected - frame_len: 828 - frame_counter: 2604 -                                                                                                                                                              frame_counter_last_valid: 2603 - resolution: 360
1697714144954: h264/aac in - frame_len: 287 - cb_current->size: 0
1697714144954: h264/aac in - error - frame size exceeds buffer size
1697714144979: h26x in - frame detected - frame_len: 2742 - frame_counter: 2607                                                                                                                                                              - frame_counter_last_valid: 2606 - resolution: 1080
1697714145005: h26x in - frame detected - frame_len: 889 - frame_counter: 2605 -                                                                                                                                                              frame_counter_last_valid: 2604 - resolution: 360
1697714145005: h26x in - frame detected - frame_len: 2846 - frame_counter: 2608                                                                                                                                                              - frame_counter_last_valid: 2607 - resolution: 1080
1697714145030: h264/aac in - frame_len: 293 - cb_current->size: 0
1697714145030: h264/aac in - error - frame size exceeds buffer size
1697714145055: h26x in - frame detected - frame_len: 875 - frame_counter: 2606 -                                                                                                                                                              frame_counter_last_valid: 2605 - resolution: 360
1697714145081: h26x in - frame detected - frame_len: 2774 - frame_counter: 2609                                                                                                                                                              - frame_counter_last_valid: 2608 - resolution: 1080
1697714145081: h264/aac in - frame_len: 290 - cb_current->size: 0
1697714145081: h264/aac in - error - frame size exceeds buffer size
1697714145106: h26x in - frame detected - frame_len: 859 - frame_counter: 2607 -                                                                                                                                     [... cut ...]
1697714199797: h26x in - frame detected - frame_len: 1102 - frame_counter: 3741 - frame_counter_last_valid: 3740 - resolution: 360
1697714199797: h264/aac in - frame_len: 1102 - cb_current->size: 49152
1697714199797: h264/aac in - frame_len: 1102 - frame_counter: 3741 - resolution: 360
1697714199797: h264/aac in - frame_write_index: 14/42
Stream detected: high res is h264, low res is h264
Failed to create RTSP server: unable to create stream socket: Address family not supported by protocol
root@Soco:~#

liebefuroscar, 2023-10-19:
I´m using your file. [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13042567/rRTSPServer.gz)
Third command only run when "RTSP alternative server" is disabled and "RTSP audio" is disabled

root@Soco:~# rRTSPServer -m y291ga -r both -a no -p 554 -d 15
1697723231638: the size of the buffer is 1786224
1697723231654: capture - mapping file fshare_frame_buf, size 1786224, to b6c3f00                                                                                                                                                             0
1697723231655: capture - closing the file fshare_frame_buf
1697723231656: capture - starting capture main loop
1697723231657: capture - buf_idx_end == buf_idx_end_prev
1697723231668: capture - ! n > 1
1697723231679: capture - ! n > 1
1697723231689: capture - warning - 3166 frame(s) lost
1697723231689: aac in - warning - 865 AAC frame(s) lost - frame_counter: 865 - f                                                                                                                                                             rame_counter_last_valid: -1
1697723231689: h264/aac in - frame_len: 282 - cb_current->size: 0
1697723231689: h264/aac in - error - frame size exceeds buffer size
1697723231714: h26x in - warning - 1149 low res frame(s) lost - frame_counter: 1                                                                                                                                                             149 - frame_counter_last_valid: -1
1697723231714: h26x in - warning - 1152 high res frame(s) lost - frame_counter:                                                                                                                                                              1152 - frame_counter_last_valid: -1
1697723231739: h264/aac in - frame_len: 275 - cb_current->size: 0
1697723231739: h264/aac in - error - frame size exceeds buffer size
1697723231765: h26x in - frame detected - frame_len: 1318 - frame_counter: 1150                                                                                                                                                              - frame_counter_last_valid: 1149 - resolution: 360
1697723231790: h26x in - frame detected - frame_len: 5465 - frame_counter: 1153                                                                                                                                                              - frame_counter_last_valid: 1152 - resolution: 1080
1697723231790: h26x in - frame detected - frame_len: 1318 - frame_counter: 1151                                                                                                                                                              - frame_counter_last_valid: 1150 - resolution: 360
1697723231815: h264/aac in - frame_len: 267 - cb_current->size: 0
1697723231815: h264/aac in - error - frame size exceeds buffer size
1697723231840: h26x in - frame detected - frame_len: 5102 - frame_counter: 1154                                                                                                                                                              - frame_counter_last_valid: 1153 - resolution: 1080
1697723231866: h26x in - frame detected - frame_len: 1373 - frame_counter: 1152                                                                                                                                                              - frame_counter_last_valid: 1151 - resolution: 360
1697723231891: h26x in - frame detected - frame_len: 4945 - frame_counter: 1155                                                                                                                                                              - frame_counter_last_valid: 1154 - resolution: 1080
1697723231891: h264/aac in - frame_len: 256 - cb_current->size: 0
1697723231891: h264/aac in - error - frame size exceeds buffer size
1697723231916: h26x in - frame detected - frame_len: 1293 - frame_counter: 1153                                                                                                                                                              - frame_counter_last_valid: 1152 - resolution: 360
1697723231916: h26x in - frame detected - frame_len: 4533 - frame_counter: 1156                                                                                                                                                              - frame_counter_last_valid: 1155 - resolution: 1080
1697723231941: h264/aac in - frame_len: 262 - cb_current->size: 0
1697723231942: h264/aac in - error - frame size exceeds buffer size
1697723231967: h26x in - frame detected - frame_len: 1294 - frame_counter: 1154                                                                                                                                                              - frame_counter_last_valid: 1153 - resolution: 360
1697723231967: h26x in - frame detected - frame_len: 4499 - frame_counter: 1157                                                                                                                                                              - frame_counter_last_valid: 1156 - resolution: 1080
1697723231999: h264/aac in - frame_len: 275 - cb_current->size: 0
1697723231999: h264/aac in - error - frame size exceeds buffer size
[... cut ...]
1697723233062: h26x in - frame detected - frame_len: 1243 - frame_counter: 1178                                                                                                                                                              - frame_counter_last_valid: 1177 - resolution: 360
1697723233062: h264/aac in - frame_len: 1243 - cb_current->size: 49152
1697723233062: h264/aac in - frame_len: 1243 - frame_counter: 1178 - resolution:                                                                                                                                                              360
1697723233062: h264/aac in - frame_write_index: 19/42
Terminated
root@Soco:~#

roleoroleo, 2023-10-19:
I'm sorry to bore you with these requests but the log is truncated.
Could you expand the window and capture the entire row?

liebefuroscar, 2023-10-19:
Dont worry. I like to help the development of the project.
"killall wd_rtsp.sh" and "killall rRTSPServer" were executed without problem.
rRTSPServer -m y291ga -r both -a no -p 554 -d 15 is the command on the first line of what I copied

roleoroleo, 2023-10-19:
The log you posted above is truncated at column 80.
I need the full log. Expand your terminal and run again the command.

liebefuroscar, 2023-10-20:
There is some point that I don't know about where it seems to go into a loop. After 5 minutes executing the command, I had to stop it manually. This is what he let me copy:
1697817492685: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492685: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 42 - counter 14337 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492685: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492703: h26x in - frame detected - frame_len: 32 - frame_counter: 14341 - frame_counter_last_valid: 14340 - resolution: 360
1697817492703: h264/aac in - frame_len: 32 - cb_current->size: 49152
1697817492703: h264/aac in - frame_len: 32 - frame_counter: 14341 - resolution: 360
1697817492703: h264/aac in - frame_write_index: 41/42
1697817492718: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492718: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 9471 - counter 14103 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492718: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492728: h264/aac in - frame_len: 280 - cb_current->size: 0
1697817492729: h264/aac in - error - frame size exceeds buffer size
1697817492729: h26x in - frame detected - frame_len: 1123 - frame_counter: 14111 - frame_counter_last_valid: 14110 - resolution: 1080
1697817492729: h264/aac in - frame_len: 1123 - cb_current->size: 262144
1697817492729: h264/aac in - frame_len: 1123 - frame_counter: 14111 - resolution: 1080
1697817492729: h264/aac in - frame_write_index: 0/42
1697817492735: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492735: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 34 - counter 14338 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492735: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492754: h26x in - frame detected - frame_len: 32 - frame_counter: 14342 - frame_counter_last_valid: 14341 - resolution: 360
1697817492754: h264/aac in - frame_len: 32 - cb_current->size: 49152
1697817492754: h264/aac in - frame_len: 32 - frame_counter: 14342 - resolution: 360
1697817492754: h264/aac in - frame_write_index: 0/42
1697817492769: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492769: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 2304 - counter 14104 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492769: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492779: h264/aac in - frame_len: 281 - cb_current->size: 0
1697817492779: h264/aac in - error - frame size exceeds buffer size
1697817492779: h26x in - frame detected - frame_len: 578 - frame_counter: 14112 - frame_counter_last_valid: 14111 - resolution: 1080
1697817492779: h264/aac in - frame_len: 578 - cb_current->size: 262144
1697817492779: h264/aac in - frame_len: 578 - frame_counter: 14112 - resolution: 1080
1697817492779: h264/aac in - frame_write_index: 1/42
1697817492785: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492785: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 30 - counter 14339 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492785: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492804: h26x in - frame detected - frame_len: 44 - frame_counter: 14343 - frame_counter_last_valid: 14342 - resolution: 360
1697817492805: h264/aac in - frame_len: 44 - cb_current->size: 49152
1697817492805: h264/aac in - frame_len: 44 - frame_counter: 14343 - resolution: 360
1697817492805: h264/aac in - frame_write_index: 1/42
1697817492819: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492819: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 2212 - counter 14105 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492819: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492830: h264/aac in - frame_len: 287 - cb_current->size: 0
1697817492830: h264/aac in - error - frame size exceeds buffer size
1697817492830: h26x in - frame detected - frame_len: 684 - frame_counter: 14113 - frame_counter_last_valid: 14112 - resolution: 1080
1697817492830: h264/aac in - frame_len: 684 - cb_current->size: 262144
1697817492830: h264/aac in - frame_len: 684 - frame_counter: 14113 - resolution: 1080
1697817492830: h264/aac in - frame_write_index: 2/42
1697817492835: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1697817492835: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 36 - counter 14340 - fMaxSize 262144 - fLimitNumBytesToStream 0
1697817492835: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817492856: capture - ! n > 1
1697817492866: h26x in - frame detected - frame_len: 23 - frame_counter: 14344 - frame_counter_last_valid: 14343 - resolution: 360
1697817492866: h264/aac in - frame_len: 23 - cb_current->size: 49152
1697817492866: h264/aac in - frame_len: 23 - frame_counter: 14344 - resolution: 360
[... cut ...]
1697817499135: VideoFramedMemorySource - doGetNextFrame() whole frame completed
1697817499136: h26x in - frame detected - frame_len: 301 - frame_counter: 14241 - frame_counter_last_valid: 14240 - resolution: 1080
1697817499136: h264/aac in - frame_len: 301 - cb_current->size: 262144
1697817499136: h264/aac in - frame_len: 301 - frame_counter: 14241 - resolution: 1080
1697817499136: h264/aac in - frame_write_index: 4/42
^Z[1]+  Stopped                    rRTSPServer -m y291ga -r both -a no -p 554 -d 15
root@Soco:~#

roleoroleo, 2023-10-20:
The log is perfect.
I don't know how to fix this problem.

closed 2024-02-18 by stale[bot]