roleoroleo/yi-hack-Allwinner-v2 #682: ONVIF PAN Left and Right are not working but TILT Up and Down yes

Zephael86, 2023-07-29:
Hi everyone,

In advance, I am not sure whether this is a Home Assistant or ONVIF issue.
Today I updated to firmware 0.2.9 on all 3 of my r30gb cameras, and observed that Pan Left and Right does not work, when executed in Home Assistant.

It works using a separate mobile application (Onvier) and using the camera's web interface.

I am using webrtc-camere and onvif integrations in Home Assistant, based on a previous guide - I think - posted here.
Even when calling Onvif Services from Developer Tools, Tilt does not have an effect.

Any idea where to continue the investigation?

`type: custom:webrtc-camera
url: rtsp://192.168.5.40/ch0_0.h264
ui: true
muted: false
poster: https://home-assistant.io/images/cast/splash.png
ptz:
  service: onvif.ptz
  data_left:
    entity_id: camera.yicam_profile_0
    pan: LEFT
  data_right:
    entity_id: camera.yicam_profile_0
    pan: RIGHT
  data_up:
    entity_id: camera.yicam_profile_0
    tilt: UP
  data_down:
    entity_id: camera.yicam_profile_0
    tilt: DOWN
`

Zephael86, 2023-07-30:
Well... as a workaround, I switched to REST calls instead of ONVIF.
As documented [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#ptz)

Added this to 'configuration.yaml'. Restart HA.

`rest_command:`
`  camera_move:`
`    url: http://{{camip}}/cgi-bin/ptz.sh?dir={{direction}}&time=0.3`
`    method: POST`
`    payload: "{{message}}"`

Then in webrtc card:

`type: custom:webrtc-camera`
`url: rtsp://192.168.5.40/ch0_0.h264`
`ui: true`
`muted: false`
`poster: https://home-assistant.io/images/cast/splash.png`
`ptz:`
`  service: rest_command.camera_move`
`  data_left:`
`    camip: 192.168.5.40`
`    direction: LEFT`
`  data_right:`
`    camip: 192.168.5.40`
`    direction: RIGHT`
`  data_up:`
`    camip: 192.168.5.40`
`    direction: UP`
`  data_down:`
`    camip: 192.168.5.40`
`    direction: DOWN`

I am closing the issue.

closed 2023-07-30 by Zephael86

roleoroleo, 2023-08-16:
I will check the onvif implementation in ha.
Thanks.