roleoroleo/yi-hack-Allwinner-v2 #687: Mqtt Changes from HA

AditaNi, 2023-08-05:
Hi,

I have seen the the HA has a new Warning.

[image]

What is hier to do?

Regards and keep up the good work.
Adi

roleoroleo, 2023-08-19:
Are you using my integration or the auto-discovery feature?

dreimer1986, 2023-08-23:
As the addin/inegration entities seem to be fine (see closed report above) we might have a problem here.

roleoroleo, 2023-08-27:
This piece of code is not mine and I don't understand exactly the new naming conventions adopted.
But please, try this script:
[mqtt_adv_homeassistant.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12448312/mqtt_adv_homeassistant.sh.gz)
Overwrite the current file in /tmp/sd/yi-hack/script/mqtt_advertise and let me know if it works.

dreimer1986, 2023-08-28:
Just checked my Home Assistant and the warning is gone now. I did nothing regarding your script yet. A recent HA update must have done something. @AditaNi do you still have the message?

dreimer1986, 2023-09-07:
Recent Beta passed by and nothing was showing up again. So I guess this one was fixed in a different way or will show up later.

Quasaro, 2023-09-07:
Hi,
just to confirm the issue is still here (HA 2023.9). Here's the log line:

`2023-09-07 16:16:08.734 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'availability_topic': 'yicam/status', 'payload_available': 'online', 'payload_not_available': 'offline', 'device': {'identifiers': ['yi-cam'], 'connections': [['mac', 'b4:c9:b9:9b:b7:9a']], 'manufacturer': 'YI', 'model': 'YI Hack', 'name': 'Yi Camera', 'sw_version': '0.2.9', 'configuration_url': 'http://192.168.2.40:8080'}, 'icon': 'mdi:network', 'state_topic': 'yicam/info_global', 'name': 'Yi Camera Hostname', 'unique_id': 'yi-cam-hostname', 'value_template': Template<template=({{ value_json.hostname }}) renders=0>, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'force_update': False, 'encoding': 'utf-8', 'availability_mode': 'latest', 'qos': 0, 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Hostname'`

Want me to test some fixes?

Thanks

Quasaro, 2023-09-07:
Hey, I've tested the script you attached and the warnings are gone!

The last warning I have is:
` Entity sensor.yi_camera_wlan_strengh (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

Thanks!

Quasaro, 2023-09-07:
Found a solution for the problem above:
Change line 235
`CONTENT='{"availability_topic":"'$MQTT_PREFIX'/'$TOPIC_BIRTH_WILL'","payload_available":"'$BIRTH_MSG'","payload_not_available":"'$WILL_MSG'","device":'$DEVICE_DETAILS','$QOS' '$RETAIN' "device_class":"signal_strength","icon":"mdi:wifi","state_topic":"'$MQTT_PREFIX'/'$MQTT_ADV_TELEMETRY_TOPIC'","name":"'$UNIQUE_NAME'","unique_id":"'$UNIQUE_ID'","value_template":"{{ ((value_json.wlan_strength|int) * 100 / 70 )|int }}","platform": "mqtt","unit_of_measurement":"%","entity_category": "diagnostic"}'`

to this

`CONTENT='{"availability_topic":"'$MQTT_PREFIX'/'$TOPIC_BIRTH_WILL'","payload_available":"'$BIRTH_MSG'","payload_not_available":"'$WILL_MSG'","device":'$DEVICE_DETAILS','$QOS' '$RETAIN' "icon":"mdi:wifi","state_topic":"'$MQTT_PREFIX'/'$MQTT_ADV_TELEMETRY_TOPIC'","name":"'$UNIQUE_NAME'","unique_id":"'$UNIQUE_ID'","value_template":"{{ ((value_json.wlan_strength|int) * 100 / 70 )|int }}","platform": "mqtt","unit_of_measurement":"%","entity_category": "diagnostic"}'`

dreimer1986, 2023-09-08:
So they just hid the error/warning... I must say that I did not check the logs, only waited for the message to appear.

Quasaro, 2023-09-08:
> So they just hid the error/warning... I must say that I did not check the logs, only waited for the message to appear.

Nope, error was still there. After thr fixes I mentioned, it was gone.

dreimer1986, 2023-09-09:
I am talking about the notification and the message in the settings menu. Both are not there anymore for me. Anyways, your fix should get this fixed and everything is fine then. :D

closed 2024-01-19 by stale[bot]