roleoroleo/yi-hack-Allwinner-v2 #691: RFUS 9.0.19.12 - Any luck with others?

leafnie, 2023-08-11:
I've been struggling to find any firmwares that support the exact pair of RFUS* + 9.0.19.12. I am unable to update further through the app.

The page says if it does not exactly match, don't try it. Is it still possible to perform this? 

This person went off the beaten path a little bit, I think.

"Also tested on Yi Camera Outdoor RFUS on firmware version 12 running 0.2.9 for h30ga and it looks good."

_Originally posted by @sirduke989 in https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/246#issuecomment-1645580875_

roleoroleo, 2023-08-21:
Please, try y21ga.
It should work.

leafnie, 2023-08-21:
I have confirmed it works with y21ga. Thanks.

closed 2023-12-19 by stale[bot]