roleoroleo/yi-hack-Allwinner-v2 #692: Clone 2 cameras

An-100, 2023-08-15:
I have Two cameras yi1080.
1.- work ok, and connect to cloud.
2.- work ok, i can reset, i can change wifi, i can found connected to mi router, put NOT connect to cloud.

Can i "clone" working camera to bad camera?  i have backup /dev/mtdblock0.bin... 7 backup 
I have some computer and linux knowledge, can I make a dd of some partitions, which ones?

both have  fw 9.0.19.02_202007141512

i can install hack-allwinner-v2 and work in "damaged" camera.

An-100, 2023-08-15:
only /dev/mtdblock6.bin and /dev/mtdblock7.bin are diferent in both cameras.

roleoroleo, 2023-08-21:
If you clone mtd6 and mtd7 you will lose the internal settings: serial number, mac address, etc...
And you won't be able to connect to the cloud because both cam will login to the cloud with the same serial.
Try to resetting and restarting from scratch the second cam.

An-100, 2023-08-21:
I have checked and the md5 from 1 to 5 and they are identical in both, I have installed your firmware and I have access to telnet, I have been able to see that the cloud command gives some errors of domains that do not exist, that is why I had assumed that when working in one and in the other not in the other the configuration or the command itself were different, I am almost 99.99% sure that it is their problem.

thanks for your time

⁣Obtener BlueMail para Android ​

An-100, 2023-08-24:
I repaired, i use mtd7 to repair mtd6, and now is paired.

Enviado el: lunes, 21 de agosto de 2023 12:02
Para: roleoroleo/yi-hack-Allwinner-v2
CC: An-100; Author

closed 2023-12-23 by stale[bot]