roleoroleo/yi-hack-Allwinner-v2 #694: Broken url for Change-WiFi-credentials in wiki

rikrdo89, 2023-08-15:
I was following these instructions to install the hack without the app:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Install-the-hack-without-the-Yi-app

However, the referred wiki page (url below) does not exist: 
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials

lenisko, 2023-08-15:
It's still there as deprecated but path changed https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)

TungstenE2, 2023-08-16:
> It's still there as deprecated, but path has changed https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)

why is depercated? Is there now way to configure wifi without original app and registering?

rikrdo89, 2023-08-16:
I am not sure why it is deprecated, but if it is not, I think someone should update the link.

For setting up wifi without having to connect to the app, I have been using this method: 
https://github.com/roleoroleo/yi-hack-Allwinner/issues/289
Just change the wifi ID and Password in the `wpa_supplicant.conf` and place it under the `Factory` folder in the SD card. I am not sure how is this method different from the "deprecated" one.

roleoroleo, 2023-08-22:
It's deprecated because you can edit wifi credentials from webui.
But it works.
Link updated.

TungstenE2, 2023-08-22:
thx, but the idea is to configure wifi without YI app, but no webui without wifi.
So for initial wifi setup this is the only way, right?
So I would not call it deprecated, this would might confuse new user.

roleoroleo, 2023-08-25:
Check the last changes I made to README and the new procedure to set wifi credentials.

closed 2023-12-23 by stale[bot]