roleoroleo/yi-hack-Allwinner-v2 #695: Yi 1080p Home BFUS - Yellow LED

lenisko, 2023-08-15:
Hey there!

BFUSY44L1U / YYS.2016

At first, I tried using y21ga, but there was no reaction (solid yellow led). After that, tried unbrick using backup files I had on microSD card, but there was no reaction for backup or any other file on empty microSD card. Any clues?

Sadly, can't remember what was the previous state of this camera.

roleoroleo, 2023-08-21:
Check the model of the cam with this procedure:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/693#discussioncomment-6779130
Then follow the unbrick guide.

lenisko, 2023-08-22:
I had to do something wrong before... All good now. Thank you!

Leaving my notes:
```
➜  yi binwalk -eM mtdblock2.bin
➜  yi cat _mtdblock3.bin.extracted/squashfs-root/homever
9.0.19.02_202007141512
```

Moved backup files to `\unbrick\y21ga` and created images.

Moved them to microSD
[image]

Camera booted successfully!

closed 2023-08-22 by lenisko