roleoroleo/yi-hack-Allwinner-v2 #698: Feature request: pass Wi-Fi credentials on the very first run of the hack

troyane, 2023-08-16:
My personal use-case was to use cameras wireless. So I need to take cameras one by one and attach to router via LAN cable, so it become visible in my network. It would be way easier if I could pass Wi-Fi credentials file alongside with the rest of the files required to run YI hack on first time.

roleoroleo, 2023-08-21:
I will add to the next release.

troyane, 2023-08-21:
Great to hear that! Thanks a lot!

troyane, 2023-09-03:
@roleoroleo There is an optional step described in [(4)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master#installation), but there is no such file in `Factory` folder of [b091qp_0.2.9.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/tag/0.2.9). Here is its content:
[image]

roleoroleo, 2023-09-08:
You are right.
I committed the code and the new readme but it's not yet released.

roleoroleo, 2023-09-08:
If you want to try it, you can add the file manually to the sd.

closed 2024-01-06 by stale[bot]