roleoroleo/yi-hack-Allwinner-v2 #699: Presets ONVIF only Preset 0 working

alfiriel, 2023-08-16:
YI Dome Camera U
h51ga_0.2.9 latest stable release
Camera Firmware 9.0.21.x

On the Webinterface I can store and use presets. The presets created via webinterface are also visible in the YI App.

But with ONVIF I am only able to use preset 0.
Checked with ONVIF Device Manager v2.2.250
When I try to go to Preset 1, I receive the error message "Die zugrunde liegende Verbindung wurde geschlossen: Die Verbindung wurde unerwartet getrennt.." English: "The underlying connection was closed: The connection was unexpectedly closed.."

Is this normal behavior? What debug logs do you need?

Thank you in advance and regards

roleoroleo, 2023-08-23:
Please send your ptz conf file:
`cat /tmp/sd/yi-hack/etc/ptz_presets.conf`

alfiriel, 2023-08-25:
What I receive if I enter the code via SSH:
```
root@cam04:~# cat /tmp/sd/yi-hack/etc/ptz_presets.conf
0=terraceDoor
1=roof
2=
3=
4=
5=
6=
7=
```

File /tmp/sd/yi-hack/etc/ptz_presets.conf is attached
[ptz_presets.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12440841/ptz_presets.zip)

alfiriel, 2023-08-28:
No authentication used

Already tried to reset the camera and do the hack again. Same behavior.

roleoroleo, 2023-09-01:
Please follow the steps below to debug application:

- Overwrite `/tmp/sd/yi-hack/www/onvif/onvif_simple_server` with this new version
[onvif_simple_server.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12497636/onvif_simple_server.gz)
- Run the command `echo 5 > /tmp/onvif_simple_server.debug` to enable verbose log
- Click "Go preset" button in ODM
- Post the content of the log `/tmp/onvif_simple_server.log` (clean personal info if present)

Run the procedure for both presets, working (0) and not working (1).

alfiriel, 2023-09-13:
[onvif_simple_server.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12600824/onvif_simple_server.log)

here you go, thank you for your help!

roleoroleo, 2023-09-23:
The log is ok.
But I fixed a little bug.
Please try this binary, overwrite your /tmp/sd/yi-hack/www/onvif/onvif_simple_server
[onvif_simple_server.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12706210/onvif_simple_server.gz)

alfiriel, 2023-09-27:
Thank you I have changed the binary. Everything seems to work now for me.
At least I have two stored working presets, which is exactly what I need.
Thank you for your time!

closed 2024-01-25 by stale[bot]