roleoroleo/yi-hack-Allwinner-v2 #701: the Free/Total Memory counter in the web interface seems wrong

nemonoid, 2023-08-16:
Example:

In the web interface it says that there are 38420KB of free RAM:

[image]

But from the SSH console, I can see that there are only 11077KB of free RAM (or 18096K available)

[image]

Is this a bug, or am I missing something ?

roleoroleo, 2023-08-23:
Free is the sum of 3 fields: free + buff/cache + available.
I don't remember why, probably I never changed this calculation after cloning yi-hack-v4.
I think it's not ok, it should be free + buff/cache.
I will fix it.

nemonoid, 2023-08-23:
> free is the sum of 3 fields: free + buff/cache + available. 

Ok, that is certainly wrong:  I just checked, and in Linux **available** is the sum of **free + the portions of buff/cache that can be released immediately**  

Good Explaination Here: https://www.howtogeek.com/456943/how-to-use-the-free-command-on-linux/

( in fact it makes sense, as the kernel uses a lot of memory for buffers and caches to speed up operations, but they are not strictly needed and can be quickly released, therefore they are considered part of the **available** memory )

So I believe the correct thing is to just quote the Available/Total Memory numbers in the Web Interface. 
So for example taking my screenshot above it would be:  _Available/Total Memory 18096/60908 KB_

roleoroleo, 2023-08-23:
I will change it.

nemonoid, 2023-08-23:
> The reason of the current calculation is the format of the output. It depends on the version, the number of columns can be different (buffers and cache columns).

Oh, I see!

Thanks

closed 2023-12-22 by stale[bot]