roleoroleo/yi-hack-Allwinner-v2 #703: delete a file button not working

jklawiter, 2023-08-18:
the delete video file button is not working. 

http://kamera/index.html?page=eventsfile&dirname=2023Y08M18D07H
there are play and delay buttons for every video file, delete button reloads page, but the file is not deleted.

I'm using latest 0.2.9 version on RFUSY62U6 1080p home camera

cannot find any live log to check if it says something… is there any?

roleoroleo, 2023-08-24:
I need the model suffix.
And please try the following steps:
- run `ls -la /tmp/sd/record/2023Y08M18D07H` and post the output
- call this url from the browser `http://kamera/cgi-bin/eventsfiledel.sh?file=2023Y08M18D07H/02M00S60.mp4` with a file name that really exists and check if the file is deleted

jklawiter, 2023-08-25:
Model suffix: y291ga

`root@YiHack:~# ls -la /tmp/sd/record/2023Y08M18D07H/
drwxr-xr-x    2 root     root         32768 Aug 18 08:05 .
drwxr-xr-x  164 root     root         32768 Aug 25 15:00 ..
-rwxr-xr-x    1 root     root         50068 Aug 18 07:08 W801M28S32.jpg
-rwxr-xr-x    1 root     root       3474940 Aug 18 07:02 W801M28S32.mp4
-rwxr-xr-x    1 root     root         48851 Aug 18 07:08 W802M01S59.jpg
-rwxr-xr-x    1 root     root       7347229 Aug 18 07:03 W802M01S59.mp4
-rwxr-xr-x    1 root     root         46619 Aug 18 07:08 W803M01S59.jpg
-rwxr-xr-x    1 root     root      10714877 Aug 18 07:04 W803M01S59.mp4
-rwxr-xr-x    1 root     root         48991 Aug 18 07:08 W804M01S59.jpg
-rwxr-xr-x    1 root     root       5285703 Aug 18 07:05 W804M01S59.mp4
-rwxr-xr-x    1 root     root         50813 Aug 18 07:16 W809M45S15.jpg
-rwxr-xr-x    1 root     root       2329300 Aug 18 07:10 W809M45S15.mp4
-rwxr-xr-x    1 root     root         53585 Aug 18 07:16 W810M01S59.jpg
-rwxr-xr-x    1 root     root       7926502 Aug 18 07:11 W810M01S59.mp4
-rwxr-xr-x    1 root     root         53330 Aug 18 07:16 W811M01S59.jpg
-rwxr-xr-x    1 root     root       8647823 Aug 18 07:12 W811M01S59.mp4
-rwxr-xr-x    1 root     root         53858 Aug 18 07:16 W812M03S57.jpg
-rwxr-xr-x    1 root     root       8603621 Aug 18 07:13 W812M03S57.mp4
-rwxr-xr-x    1 root     root         48434 Aug 18 07:20 W813M01S59.jpg
-rwxr-xr-x    1 root     root       5110308 Aug 18 07:14 W813M01S59.mp4
-rwxr-xr-x    1 root     root         51101 Aug 18 07:24 W819M10S50.jpg
-rwxr-xr-x    1 root     root       8631383 Aug 18 07:20 W819M10S50.mp4
-rwxr-xr-x    1 root     root         56381 Aug 18 07:24 W820M01S59.jpg
-rwxr-xr-x    1 root     root       8795722 Aug 18 07:21 W820M01S59.mp4
-rwxr-xr-x    1 root     root         59165 Aug 18 07:28 W821M01S59.jpg
-rwxr-xr-x    1 root     root      10666944 Aug 18 07:22 W821M01S59.mp4
-rwxr-xr-x    1 root     root         56206 Aug 18 07:28 W822M01S59.jpg
-rwxr-xr-x    1 root     root      10145679 Aug 18 07:23 W822M01S59.mp4
-rwxr-xr-x    1 root     root         47838 Aug 18 07:28 W823M01S59.jpg
-rwxr-xr-x    1 root     root       5071006 Aug 18 07:24 W823M01S59.mp4
-rwxr-xr-x    1 root     root         44845 Aug 18 07:48 W839M20S40.jpg
-rwxr-xr-x    1 root     root       7426960 Aug 18 07:40 W839M20S40.mp4
-rwxr-xr-x    1 root     root         60807 Aug 18 07:48 W840M01S59.jpg
-rwxr-xr-x    1 root     root       9900028 Aug 18 07:41 W840M01S59.mp4
-rwxr-xr-x    1 root     root         56808 Aug 18 07:49 W841M01S59.jpg
-rwxr-xr-x    1 root     root       5515703 Aug 18 07:42 W841M01S59.mp4
-rwxr-xr-x    1 root     root         44562 Aug 18 07:49 W844M54S06.jpg
-rwxr-xr-x    1 root     root        598734 Aug 18 07:45 W844M54S06.mp4
-rwxr-xr-x    1 root     root         45132 Aug 18 07:53 W845M02S58.jpg
-rwxr-xr-x    1 root     root       5984732 Aug 18 07:46 W845M02S58.mp4
-rwxr-xr-x    1 root     root         40502 Aug 18 07:53 W846M02S58.jpg
-rwxr-xr-x    1 root     root       9303004 Aug 18 07:47 W846M02S58.mp4
-rwxr-xr-x    1 root     root         46114 Aug 18 07:53 W847M01S59.jpg
-rwxr-xr-x    1 root     root      11770206 Aug 18 07:48 W847M01S59.mp4
-rwxr-xr-x    1 root     root         45840 Aug 18 07:53 W848M01S59.jpg
-rwxr-xr-x    1 root     root       8537642 Aug 18 07:49 W848M01S59.mp4
-rwxr-xr-x    1 root     root         46674 Aug 18 08:05 W858M10S50.jpg
-rwxr-xr-x    1 root     root       7342760 Aug 18 07:59 W858M10S50.mp4
-rwxr-xr-x    1 root     root         45918 Aug 18 08:05 W859M01S59.jpg
-rwxr-xr-x    1 root     root       4848432 Aug 18 08:00 W859M01S59.mp4
`
About the URL calling yours one gives JSON error (error: "false")
changing the filename to `http://kamera/cgi-bin/eventsfiledel.sh?file=2023Y08M18D07H/W859M01S59.mp4` shows error = "true" but the file is not deleted:

`root@YiHack:~# ls -la /tmp/sd/record/2023Y08M18D07H/ | tail -n 2
-rwxr-xr-x    1 root     root         45918 Aug 18 08:05 W859M01S59.jpg
-rwxr-xr-x    1 root     root       4848432 Aug 18 08:00 W859M01S59.mp4
`

roleoroleo, 2023-08-28:
> About the URL calling yours one gives JSON error (error: "false")

Sorry for the misunderstaing: error false means success.

Your cam uses a different naming convention.
Try to change these lines to the file /tmp/sd/yi-hack/www/httpd/cgi-bin/eventsfiledel.sh

Before:
```
    if [ "M${1:17:1}" != "MM" ] ; then
        FILE="none"
    fi
    if [ "S${1:20:1}" != "SS" ] ; then
        FILE="none"
    fi
```

After:
```
    if [ "M${1:17:1}" != "MM" ] && [ "M${1:19:1}" != "MM" ] ; then
        FILE="none"
    fi
    if [ "S${1:20:1}" != "SS" ] && [ "S${1:22:1}" != "SS" ] ; then
        FILE="none"
    fi
```

jklawiter, 2023-08-28:
> Sorry for the misunderstaing: error false means success.
yes, I understood it a moment after sending the message, but I didn't want to write something that doesn't add anything anyway

location is also little different here:
$ pwd
/tmp/sd/yi-hack/www/cgi-bin

Anyway after this change both
calling: http://kamera/cgi-bin/eventsfiledel.sh?file=2023Y08M25D19H/E227M55S05.mp4
and click delete button on http://kamera/?page=eventsfile&dirname=2023Y08M25D19H
works. 

What have I to do to keep it working after next firmware update?

roleoroleo, 2023-09-02:
> What have I to do to keep it working after next firmware update?

Nothing.
This change will be part of the next fw udpate.

closed 2024-01-01 by stale[bot]