roleoroleo/yi-hack-Allwinner-v2 #705: Kami Baby support

osh123, 2023-08-21:
Any option to have Kami (YI) Baby camera support?
Serial number is starting with LFUS, firmware version is 1.3.0.0C.

osh123, 2023-09-03:
Hi @roleoroleo,

Thanks for the reply, I've ran backup script - it did work, can I post it here? does it contain any sensitive information of my Kami/YI account?

Thanks!

roleoroleo, 2023-09-08:
Ok, you can remove the link, if you want.
The cam seems to be Allwinner based and probably compatible with this hack:
https://github.com/roleoroleo/yi-hack-Allwinner

But I need some time to add the support.
I will create a beta version to try.
Please, open the issue in the repo https://github.com/roleoroleo/yi-hack-Allwinner

roleoroleo, 2023-09-08:
How long ago did you buy it?

osh123, 2023-09-08:
> Ok, you can remove the link, if you want. The cam seems to be Allwinner based and probably compatible with this hack: https://github.com/roleoroleo/yi-hack-Allwinner
>
> But I need some time to add the [...]

Will do, thank you!

> How long ago did you buy it?

10 months ago.

closed 2024-01-06 by stale[bot]