roleoroleo/yi-hack-Allwinner-v2 #707: Motion detection without WiFi?

drewsed, 2023-08-22:
Hi, today in the morning I turned on my YiHome 1080p camera out of the range of my wifi to record offline on motion detection events.

In the evening I turned on my mobile hotspot (android) with the "home wifi ssid and pw" to get the files from the camera. I did not get a connection due to not finding the ip and or restrictions on cross talk within the mobile hotspot.

then I took the cam home, and accesed the webserver to download the files.

there was only a video of me walking in front of the camera doing the mobile hotspot thing.

Does the cam properly start without wifi and is it intended to record on motion in offline situations?

I was never thinking about this and used the camera offline for the first time.

Now i wonder if there has to be a internet connection or only a wifi ap to get it to work. 

I hacked the camera in around 2019 and since then I only used the rtsp and onviv features.

Thanks in advance. I didn't find any related issues, sorry if they exist.

roleoroleo, 2023-08-25:
This is a know problem.
Yi boot program doesn't run processes if the cam is not connected.
I don't know how to fix it.

drewsed, 2023-08-25:
Thank you very much. Do you know if I can boot it with wifi ap connection and then disconnect to record in offline mode? is internet required or just wifi?

closed 2023-08-25 by drewsed