roleoroleo/yi-hack-Allwinner-v2 #716: Blue light keep blinking

jichay, 2023-09-13:
Hardware: QFUS
Version: 9.0.35.*

Hello,
So, I just flashed the new firmware to my Yi cam (with the latest twist from the main branch for the default WiFi configuration), but after 1 hour, it was still flashing blue. 
So, I decided to unplug it and check the files on the SD card. It seems to have worked; the `Factory.done` folder is present. 
But powering it back on was not doing anything except blinking blue.

roleoroleo, 2023-09-23:
If you want to use the configuration feature for wifi connection, use the last release and add/overwrite the files from https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/sdhack/y291ga/Factory

Phorik, 2023-11-07:
Hi,
I have the same HW/FW and the same problem. Used the latest y291ga release.
Blue light flashing and no wifi connection.
Do i need a special format after "wifi_ssid=" and "wifi_psk="?
Like, keep the $ after the key or something?

roleoroleo, 2023-11-08:
No special format.
Does the cam speak asking for network configuration?

Phorik, 2023-11-10:
no, it's not saying anything, just blue blinking light and no connection to the router

roleoroleo, 2023-11-11:
Try to reset the cam with the button.

closed 2024-03-10 by stale[bot]

weibell, 2024-12-28:
Hi @Phorik, I was experiencing the same issue and was able resolve it, here are the steps I took:

- Extract version 0.3.3 to the SD card and edit Factory/configure_wifi.cfg as described in the [install procedure, step 4](https://github.com/roleoroleo/yi-hack-Allwinner-v2?tab=readme-ov-file#install-procedure)
- When powering on the camera, observe that the blue light keeps on blinking, even when powering it on again
- Hold the reset button for a few seconds, the camera now says "waiting to connect" when powering on
- Follow the steps described in [Change WiFi credentials (deprecated)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated)):
> Download the following archive [sdwifi.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/15275311/sdwifi.tar.gz) and decompress it to a FAT32 formatted SD card in the folder "yi-hack"; don't clean the SD card, it must contain both the hack (already applied) and this decompressed archive.
> Edit the file configure_wifi.cfg according to your needs. Pay attention to the special chars:
> ...
> When you open the SD card you will find these new files in the yi-hack folder.
> / --- yi-hack --- configure_wifi.cfg
>               |-- configure_wifi.sh
>               |-- startup.sh
> Insert the SD into the cam and power on it.
> The cam will reboot once.
> Check the wifi connection.
> Delete the files listed above (configure_wifi.cfg, configure_wifi.sh and startup.sh).
- The camera now successfully connects to the network